react-native-camera/react-native-camera #2853: Facing hardship in installing this

Return-1, 2020-06-04:
This is relating to DX. Getting multiple barriers while trying to install as a first time user.

It was impossible to set up without following **https://react-native-community.github.io/react-native-camera/docs/migrationv2** which i was directed to by another developer.

It works fine for me now but my advice is please attempt to install this in a vanilla react-native fresh install to feel the struggle of someone installing v2 directly.

Just putting this here as friendly feedback, it works for me now so there's no agenda.

sibelius, 2020-06-04:
can you help improve docs to improve the DX?

Return-1, 2020-06-05:
Not at this time unfortunately. I would have to come back to it later in time and see what would actually need to be put forth as i'm not a regular maintainer or familiar with all the ways people use the module, i'm simply reporting this.

It is however -for what it's worth- my and other's experience that the installation on a modern post 0.60 RN version requires more than meets the eye, looking at previous versions migration guides and so on. Not any complex setup mind you, plain ol basic fresh install. This feels like an important issue that could swiftly be resolved by someone who knows this module far better than I.

Thank you for the response. I will attempt to come back to it in time but i don't want to promise and subsequently block this.

closed 2020-08-11 by stale[bot]