react-native-camera/react-native-camera #2860: playSoundOnCapture implementation for iOS

airowe, 2020-06-05:
# Question

**To Do First**
- [ ] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [ ] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [ ] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**
Is there a `playSoundOnCapture` implementation for iOS?

**Tags**
ios sound

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

0xori, 2020-08-05:
is it possible?

0xori, 2020-10-12:
still happening

closed 2021-01-13 by stale[bot]