react-native-camera/react-native-camera #2867: App Crash on Camera with Fatal Exception

ANIBIT14, 2020-06-09:
Hi, in production environment  in our app on Android  - we are getting a crash which is : -

Fatal Exception: java.lang.IllegalArgumentException
Invalid rotation=-152
android.hardware.Camera$Parameters.setRotation

Fatal Exception: java.lang.IllegalArgumentException: Invalid rotation=-152
       at android.hardware.Camera$Parameters.setRotation(Camera.java:3737)
       at com.google.android.cameraview.Camera1.adjustCameraParameters(Camera1.java:119)
       at com.google.android.cameraview.Camera1$2.run(Camera1.java:22)
       at android.os.Handler.handleCallback(Handler.java:883)
       at android.os.Handler.dispatchMessage(Handler.java:100)
       at android.os.Looper.loop(Looper.java:214)
       at android.os.HandlerThread.run(HandlerThread.java:67)

If anybody have any idea why this may be happening, please share it would be of great help.

fabriziobertoglio1987, 2020-06-15:
you can not set negative rotation to camera?

ANIBIT14, 2020-06-15:
HI, @fabriziobertoglio1987, thanks for the reply but the thing is I have not set negative rotation for the camera in react-native code, I just used the official doc to make camera config, could it be some problem due to my config or the way I am using React Native Camera

fabriziobertoglio1987, 2020-06-15:
here is a couple of places where rotation is set, so avoid using those functionalities and tell us if problem persists

https://github.com/react-native-community/react-native-camera/search?q=setRotation&unscoped_q=setRotation

if it does not we found the cause of the problem

additionally give a read to https://www.google.com/search?client=ubuntu&channel=fs&q=react+native+camera+rotation&ie=utf-8&oe=utf-8

and try to apply those solutions, maybe they are settings automatically the wrong rotation but you can over ride that

ANIBIT14, 2020-06-15:
Okay, thanks for the help, will check with these

hamzaouhssain1993, 2020-06-24:
I am getting the same an invalid set rotation. this appear only in Samsung Galaxy Tab A (2016) and only in the release it has happened so this is the only information we got about the problem.
`Fatal Exception: java.lang.IllegalArgumentException: Invalid rotation=-42
       at android.hardware.Camera$Parameters.setRotation(Camera.java:3210)
       at com.google.android.cameraview.Camera1.setDeviceOrientation(Camera1.java:895)
       at com.google.android.cameraview.CameraView$1.onDisplayOrientationChanged(CameraView.java:142)
       at com.google.android.cameraview.DisplayOrientationDetector.dispatchOnDisplayOrientationChanged(DisplayOrientationDetector.java:114)
       at com.google.android.cameraview.DisplayOrientationDetector$1.onOrientationChanged(DisplayOrientationDetector.java:90)
       at android.view.OrientationEventListener$SensorEventListenerImpl.onSensorChanged(OrientationEventListener.java:143)
       at android.hardware.SystemSensorManager$SensorEventQueue.dispatchSensorEvent(SystemSensorManager.java:964)
       at android.os.MessageQueue.nativePollOnce(MessageQueue.java)
       at android.os.MessageQueue.next(MessageQueue.java:325)
       at android.os.Looper.loop(Looper.java:142)
       at android.app.ActivityThread.main(ActivityThread.java:7000)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:441)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1408)`

kevin-renier, 2020-06-25:
> Okay, thanks for the help, will check with these

@ANIBIT14 did you find any clue? Any particular device ? 

I join @hamzaouhssain1993, only happening on Samsung Galaxy Tab A (2016) on our side too (on 3000 users with many different devices). Cannot reproduce it on another Samsung Galaxy Tab model than this very particular one unfortunately :(

ANIBIT14, 2020-06-25:
@kevin-renier it hasn't happened for us again in the last week, after I added writeExif: { Orientation: 6 }, this in takePictureAsync(options) ... options which are const options = {
      base64: true,
      orientation: 'portrait',
      width: 800,
      height: 800,
      fixOrientation: true,
      forceUpOrientation: true,
      writeExif: { Orientation: 6 },
      mirrorImage: true,
    };
I found about adding this property while I was searching for solutions on net.
will surely update if this crash occurs again's on a device.

kevin-renier, 2020-08-04:
> will surely update if this crash occurs again's on a device.

Hi @ANIBIT14, can you confirm this fix is still working 100% and can be applied?

ANIBIT14, 2020-08-04:
@kevin-renier haven't got the crash, since the day I added the writeExif : { Orientation: 6 } as mentioned above. I monitor crashlytics on an hourly basis, can say didn't happened in android after that.

kevin-renier, 2020-08-05:
Test

ANIBIT14, 2020-08-29:
Git this error again after a long time @kevin-renier so keep eyes open for crashes if you are using rncamera:
It was on a Moto G5 Plus (XT1686) - Android 10
Fatal Exception: java.lang.IllegalArgumentException: Invalid rotation=-116
       at android.hardware.Camera$Parameters.setRotation(Camera.java:3730)
       at com.google.android.cameraview.Camera1.a(Camera1.java:41)
       at com.google.android.cameraview.CameraView$1.onDisplayOrientationChanged(CameraView.java:11)
       at com.google.android.cameraview.DisplayOrientationDetector.a(DisplayOrientationDetector.java:4)
       at com.google.android.cameraview.DisplayOrientationDetector$1.onOrientationChanged(DisplayOrientationDetector.java:86)
       at android.view.OrientationEventListener$SensorEventListenerImpl.onSensorChanged(OrientationEventListener.java:143)
       at android.hardware.SystemSensorManager$SensorEventQueue.dispatchSensorEvent(SystemSensorManager.java:837)
       at android.os.MessageQueue.nativePollOnce(MessageQueue.java)
       at android.os.MessageQueue.next(MessageQueue.java:336)
       at android.os.Looper.loop(Looper.java:174)
       at android.app.ActivityThread.main(ActivityThread.java:7356)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:491)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:930)

theasc, 2020-09-10:
I'm also getting several crashes on a Samsung galaxy tab A (android 8.1.0), here's the stack: 
`Fatal Exception: java.lang.IllegalArgumentException: Invalid rotation=-226
       at android.hardware.Camera$Parameters.setRotation(Camera.java:3210)
       at com.google.android.cameraview.Camera1.setDeviceOrientation(Camera1.java:895)
       at com.google.android.cameraview.CameraView$1.onDisplayOrientationChanged(CameraView.java:142)
       at com.google.android.cameraview.DisplayOrientationDetector.dispatchOnDisplayOrientationChanged(DisplayOrientationDetector.java:114)
       at com.google.android.cameraview.DisplayOrientationDetector$1.onOrientationChanged(DisplayOrientationDetector.java:90)
       at android.view.OrientationEventListener$SensorEventListenerImpl.onSensorChanged(OrientationEventListener.java:143)
       at android.hardware.SystemSensorManager$SensorEventQueue.dispatchSensorEvent(SystemSensorManager.java:964)
       at android.os.MessageQueue.nativePollOnce(MessageQueue.java)
       at android.os.MessageQueue.next(MessageQueue.java:325)
       at android.os.Looper.loop(Looper.java:142)
       at android.app.ActivityThread.main(ActivityThread.java:7000)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:441)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1408)`

closed 2021-01-13 by stale[bot]