react-native-camera/react-native-camera #2870: Build conflict

kelvin-dev, 2020-06-12:
Latest source on the master branch gets conflicted with below react-native extension
[](https://github.com/Agontuk/react-native-geolocation-service)
at 
```ext {
  compileSdkVersion   = 28
  buildToolsVersion   = "28.0.3"
  minSdkVersion       = 16
  targetSdkVersion    = 28

  // Any of the following will work
  **googlePlayServicesVersion      = "17.0.0"**
  // playServicesVersion         = "17.0.0"
  // playServicesLocationVersion = "17.0.0"
}```
**Build logs:**
``FAILURE: Build failed with an exception.

* What went wrong:
Could not determine the dependencies of task ':react-native-camera:compileGeneralDebugAidl'.
> Could not resolve all task dependencies for configuration ':react-native-camera:generalDebugCompileClasspath'.
   > Could not find com.google.android.gms:play-services-vision:17.0.0.
     Required by:
         project :react-native-camera

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 13s```

Atww, 2020-06-16:
same problem

dotdotgod, 2020-06-19:
try to set version "17.0.2"?

leononly, 2020-07-03:
same problem to me too

closed 2020-09-08 by stale[bot]

jmlavoier, 2020-09-16:
Same problem here, has anyone solved the problem, I didn't find anything in the documentation about this troubleshooting. It happens when we are using the `react-native-geolocation-service` with `react-native-camera`. That's why that geolocation library requires [googlePlayServicesVersion = "17.0.0"](https://github.com/Agontuk/react-native-geolocation-service#android) as described in documentation. And I've googled about this version and I found solutions to change the `googlePlayServicesVersion` to `17.0.2`. But it causes problems in that geolocation library.

I'm thinking this isn't about the documentation problem. It could be about compatibility.

jmlavoier, 2020-09-16:
I solved this with this refactor in [documentation about modify buildgradle](https://github.com/react-native-community/react-native-camera/blob/master/docs/installation.md#modifying-buildgradle)

In the doc, we don't have a good explanation about that modification. When and why I should do it? 

If someone is getting the same error, changing `missingDimensionStrategy 'react-native-camera', 'general'` to `missingDimensionStrategy 'react-native-camera', 'mlkit'` worked to me.

leoceolin, 2020-10-30:
> I solved this with this refactor in [documentation about modify buildgradle](https://github.com/react-native-community/react-native-camera/blob/master/docs/installation.md#modifying-buildgradle)
>
> In [...]

not work for me !

my android/build.gradle file:

        buildToolsVersion = "29.0.2"
        minSdkVersion = 16
        compileSdkVersion = 29
        targetSdkVersion = 29
        supportLibVersion = '28.0.0'
        googlePlayServicesVersion = "17.0.0"

my android/app/build.gradle

missingDimensionStrategy 'react-native-camera', 'mlkit'

need help !

rorlf, 2021-01-08:
I'm getting the same error.

tnghia944, 2021-06-14:
ADD in  your build.gragle
       googlePlayServicesVersion  = "17.0.0"
        googlePlayServicesVisionVersion = "17.0.2"