react-native-camera/react-native-camera #2872: Why is three a half second delay between barcode scans?

manishkumar03, 2020-06-13:
# Question

**To Do First**
- [ ] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [ ] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [ ] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**
There is a check `timePassedSinceSubmittingForBarcode > 0.5` in the code which basically means that barcode detection will be attempted once every 0.5 seconds. This limits the camera performance and causes delays in scanning. I wanted to know if there is any historical reason for this. If not, I'd like to change it. The same check is applicable for text and face detection.

**Tags**

Performance

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

sibelius, 2020-06-13:
Let’s reduce this, or make this configurable

fabriziobertoglio1987, 2020-06-15:
@sibelius this could happen. This could be solved in my barcode scanner refactoring in https://github.com/react-native-community/react-native-camera/pull/2766 

My logic would decode the Image after turning it in the correct orientation (I still need to add some logic to handle all camera orientations).

https://github.com/react-native-community/react-native-camera/blob/5abbacaac4448311775c5544ea5feb3c3b5b94f9/android/src/main/java/org/reactnative/camera/tasks/BarCodeScannerAsyncTask.java#L39-L63

while the previous logic would 

1) Try to decode the Barcode in the original camera orientation

https://github.com/react-native-community/react-native-camera/blob/7706ccbea7cb53d4573ff93e7308cccf85d77fb9/android/src/main/java/org/reactnative/camera/tasks/BarCodeScannerAsyncTask.java#L76-L87

2) If the Barcode could not be decoded (because the image was in the wrong orientation), catch `NotFoundException` and try again decoding it

https://github.com/react-native-community/react-native-camera/blob/7706ccbea7cb53d4573ff93e7308cccf85d77fb9/android/src/main/java/org/reactnative/camera/tasks/BarCodeScannerAsyncTask.java#L88-L100

3) Repeat 2)

https://github.com/react-native-community/react-native-camera/blob/7706ccbea7cb53d4573ff93e7308cccf85d77fb9/android/src/main/java/org/reactnative/camera/tasks/BarCodeScannerAsyncTask.java#L102-L114

4) Repeat 2)

https://github.com/react-native-community/react-native-camera/blob/7706ccbea7cb53d4573ff93e7308cccf85d77fb9/android/src/main/java/org/reactnative/camera/tasks/BarCodeScannerAsyncTask.java#L115-L126

5) End

https://github.com/react-native-community/react-native-camera/blob/7706ccbea7cb53d4573ff93e7308cccf85d77fb9/android/src/main/java/org/reactnative/camera/tasks/BarCodeScannerAsyncTask.java#L127-L129

My current policy is only writing Pull Request if an Issue already exist, seems that this justifies completion of pull request https://github.com/react-native-community/react-native-camera/pull/2766 

but I never experienced delay of 3 seconds (I never tested catch statement at point 4)

@sibelius what is your honest opinion? I am searching for an issue to solve on react-native. I reached [page 6](https://github.com/facebook/react-native/issues?page=6&q=is%3Aissue+is%3Aopen+sort%3Areactions-%2B1-desc) and ended up here little bit demoralised :smile:

sibelius, 2020-06-15:
A think the simple fix for this issue would be to remove the delay or make it customizable

About fix barcode to scan in any degree we can do that, not sure how cost this approach is

fabriziobertoglio1987, 2020-06-15:
@sibelius 

The code above is built to scan on all 4 orientations, but it not optimized. 
Instead of doing `try` ... `catch` four times we should 

use the rotation 

https://github.com/react-native-community/react-native-camera/blob/04e6f0674c7f8850d302194df1bd7d39162727e0/android/src/main/java/org/reactnative/camera/RNCameraView.java#L157-L158

pass it to the `BarCodeScannerAsyncTask` constructor
https://github.com/react-native-community/react-native-camera/blob/04e6f0674c7f8850d302194df1bd7d39162727e0/android/src/main/java/org/reactnative/camera/RNCameraView.java#L175

rotate the image in the correct direction before scanning
https://github.com/react-native-community/react-native-camera/blob/5abbacaac4448311775c5544ea5feb3c3b5b94f9/android/src/main/java/org/reactnative/camera/tasks/BarCodeScannerAsyncTask.java#L45

Thanks. Fabrizio

sibelius, 2020-06-15:
what about trying to decode in parallel in 4 rotations?

manishkumar03, 2020-06-15:
Are you sure the barcode scanner does not scan automatically in all orientations? I was under the impression that native APIs already did that. Please correct me if I am wrong.

fabriziobertoglio1987, 2020-06-16:
@sibelius No. If by parallel you mean running the same image barcode decoding in 4 different threads for each orientation. Executing this calculation almost in parallel as background task.

I think it is pretty complex task to run it in Parallel on different thread, so I would not discuss about that. `BarCodeScannerAsyncTask` is `async` but as you can see from the quoted above code the 4 directions are handled consequentially in the `try`...`catch` statement. Just set a breakpoint on each of those statements and see them executing, or `Log.w` and see that they run the statement but also return that `NotFoundError`. 

 @manishkumar03 yes, that is what I stated in the above comment. Sorry but I'll not be replying messages for a while, I need to work on other projects now. My advice to you is writing your own Native Camera Component forking this or [react-native-camera-kit](https://github.com/wix/react-native-camera-kit), open source it and hopefully they will help you mantain it. Thanks a lot. Fabrizio

sibelius, 2020-06-16:
@manishkumar03 remove the delay, test

and if it is good, we can merge