react-native-camera/react-native-camera #2873: Silent Crash on ios

interstates21, 2020-06-16:
ios crashes with no feedback on either Navigating to CameraScreen or accepting permission request on CameraScreen for the first time.

react-native-camera: 3.30.0

Permissions: 
	<key>NSCameraUsageDescription</key>
	<string>$(PRODUCT_NAME) would like to use your camera in order to make the inspection process images</string>
	<key>NSPhotoLibraryAddUsageDescription</key>
	<string>$(PRODUCT_NAME) would like to save photos to your photo gallery in order to upload the inspection process images </string>
	<key>NSMicrophoneUsageDescription</key>
	<string>$(PRODUCT_NAME) would like to use your microphone to launch RNCamera </string>
	<key>NSPhotoLibraryUsageDescription</key>
	<string>$(PRODUCT_NAME) would like access to your photo gallery in order to upload the inspection process images</string>

interstates21, 2020-06-18:
The issue arises after adding to the Podfile
```
 pod 'react-native-camera', path: '../node_modules/react-native-camera', subspecs: [
  'BarcodeDetectorMLKit'
  ]
```

interstates21, 2020-06-19:
The issue appeared because BarcodeDetector requires firebase integration which was not clear from the documentation

closed 2020-06-19 by interstates21

luism3861, 2020-07-08:
> The issue arises after adding to the Podfile
> 
> ```
>  pod 'react-native-camera', path: '../node_modules/react-native-camera', subspecs: [
>   'BarcodeDetectorMLKit'
>   ]
> ```

it works for me, delete. this line if you have using barcode you don't need this, for me it works you only need permissions in `Info.plist` and call prop in component `onBarCodeRead`