react-native-camera/react-native-camera #2875: Error scanning QRs

mlopezfernandez, 2020-06-18:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues? 

**Platforms**

 

**Versions**

- Android: 10, 9, 7
- react-native-camera: 3.30.0
- react-native: 0.62.2
- react: 16.11.0
  

**Description/Current Behaviour**

When the camera is opened for QR scanning an error pops up on the screen. We realized that when "onGoogleVisionBarcodesDetected" prop is removed this error doesn't appear.

[image]

This error only happens on Android 9 and 10.

**Steps to Reproduce**
Open camera in Android 9 or 10 for scanning QR

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

gablorquet, 2020-06-19:
It was working fine on our client devices and we started getting errors since an Android update this week, maybe it has something to do with it ?

jbauer-sds, 2020-06-22:
Experiencing this as well when using onGoogleVisionBarcodesDetected for both QR and 128.

brendan1809, 2020-06-23:
i tried console log onGoogleVisionBarcodesDetected it will always trigger it with unknown format even i am not scanning any qr and barcode

adeshpandey, 2020-06-23:
Happening with me too, even onBarCodeRead returns partial qr code

mlopezfernandez, 2020-06-24:
@adeshpandey What do you mean with partial qr code? We are trying to use this prop but the returned object is undefined.

adeshpandey, 2020-06-24:
Hi @mlopezfernandez 
My QR Code is: https://imgur.com/AqUySJ8 . RNCamera is returning: ^8r� but full QR code content is: ^8r�G0E :v(wp$H���ك��/�b��DH\�T�F!��g�[@�����Q
��S���ȇp1�~�J�~��IEtIe�E�

I have verified with many other tools.

adeshpandey, 2020-06-24:
Hi @mlopezfernandez 
Anybody, who has solved this ? I have even tried to look into the source code but seems nothing helpful. I have tried to base64 encode the rawBytes in the BarCodeReadEvent.java and decode in my component but still no luck.

gablorquet, 2020-06-24:
I had to add a check for 'UNKNOWN_FORMAT' type before invoking my callback function so, more a workaround than an actual fix but it works

jbauer-sds, 2020-06-24:
Hi @gablorquet, your post has given me a bit of hope. Could you elaborate a bit? Which file did you add the 'UNKNOWN_FORMAT' type to?

gablorquet, 2020-06-24:
Oh I added it directly to the function I pass to onGoogleVisionBarcodesDetected

jbauer-sds, 2020-06-24:
Here's what I have for my onGoogleBarcodesDetected: 

[image]

Unfortunately, even though I have specified `barCodeTypes={[RNCamera.Constants.BarCodeType.qr]}` I am getting a TON of scans with the format of None. This seems to be degrading performance a bit but you are right, this workaround does work.

Thanks @gablorquet!

haikov, 2020-06-26:
The same issue started happening to my app in production as well about a week ago, which made the app useless for some chunk of users. As I've found out, camera module started calling `onGoogleVisionBarcodesDeceted` with "barcodes" of types `UNKNOWN_FORMAT`, right after opening the camera.

As a hot fix, I've added the following check inside the `onGoogleVisionBarcodesDeceted` callback, which fixed errors, but the root cause is still there - the function is being called all the time:
```
if (barcodes.length > 0 && barcodes[0].type === 'UNKNOWN_FORMAT') {
  return;
}
```

Maybe someon have any idea what happened and why is it started to behave this way? I didn't update the app for a month in the Play Store, so it happened due to some over-the-air library update or maybe Android? Also, would be great to find the proper fix for that, so callback won't be fired all the time.

jbauer-sds, 2020-06-27:
So far it only happens on Google Play Services 20.21.17 for me (040306-316502805).

mlopezfernandez, 2020-06-29:
@adeshpandey We are using @gablorquet's workaround. @gablorquet thanks for this solution until Google fix this bug.

closed 2020-09-05 by stale[bot]

zedart, 2020-12-17:
Use onBarCodeRead instead.
        onBarCodeRead={this.barcodeRecognized.bind(this)}
  barcodeRecognized = (e) => {
      console.warn(JSON.stringify(e)); // you can show only e.data 
      this._onSendBarCode('12557', e.data);
  };

kstoykov, 2021-01-16:
@zedart 's solution works perfectly