react-native-camera/react-native-camera #2876: Width and height values swapped in TakePictureResponse

youn123, 2020-06-22:
# Bug Report

**To Do First**

- [Y] Did you try latest release?
- [N] Did you try master?
- [Y] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: 10
- react-native-camera: 3.30
- react-native: 0.62.2
- react: 16.11.0

**Description/Current Behaviour**
I want to take a standard 4:3 aspect ratio picture. I pass in `ratio='4:3'` to `RNCamera` as part of its prop. However, when I do `takePictureAsync`, the `width` and `height` returned in `TakePictureResponse` are 1440 and 1080, respectively. I expect them to be the other way around (`width` to be 1080, and `height` to be 1440).

**Expected Behaviour**
Described above.

**Steps to Reproduce**
Set up `RNCamera` component with `ratio` props set. Do `TakePictureAsync` and read the returned `width` and `height` values.

renatobenks, 2020-06-22:
what do you think @sibelius?

cristianoccazinsp, 2020-06-24:
Are you forcing up orientation or using exif orientation? That might explain it.  Even if you took the picture in portrait, it will most likely be in landscape internally.

You should be using `fixOrientation` and/or `forceUpOrientation` to get the desired result. Have you looked at the orientation prop that is also returned?

youn123, 2020-06-24:
Setting `fixOrientation` and `forceUpOrientation` seems to have fixed the issue. Good to know that pictures taken in portrait is internally landscape. Thank you for your help.

closed 2020-08-30 by stale[bot]