react-native-camera/react-native-camera #2878: scans only a part of whole bar code

adeshpandey, 2020-06-24:
# Bug Report

**To Do First**

- [ ] Did you try latest release?
- [ ] Did you try master?
- [ ] Did you look for existing matching issues?

**Platforms**

**Versions**

- Android:
- iOS:
- react-native-camera:
- react-native:
- react:
  

**Description/Current Behaviour**

**Expected Behaviour**

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

My QR Code is: https://imgur.com/AqUySJ8 . RNCamera is returning: ^8r� but full QR code content is: ^8r�G0E :v(wp$H���ك��/�b��DH\�T�F!��g�[@�����Q
��S���ȇp1��J���IEtIe�E�

I have verified with many other tools. I have tried to raise my concern on another bug but I realised that that bug is related to google bar code scanner which is also an issue in the package.

adeshpandey, 2020-06-24:
Hi @sibelius 
May I know what are the guidelines to mark an issue to feature request? Since the concern I raised here seems to be a bug to me but you have labelled it to Feature.

closed 2020-06-25 by adeshpandey

adeshpandey, 2020-06-25:
Seems the qr scanner iamge issue so I removed this package from my project and used some custom package to achieve this.

sibelius, 2020-06-25:
which one?

adeshpandey, 2020-06-25:
Hi @sibelius 
I have dig more and found that the text is getting truncated in the react view. BarCodeReadEvent.java was returning valid text but in the react-native view it was getting truncated. I have modified module according to my need to return base64 encoded string from the BarCodeReadEvent which is solving my problem.

My custom package also had same issues so I jumped back to this module and working with base64 which will be decoded in my API.