react-native-camera/react-native-camera #2880: Even i gave audio permissions in RNCamera getting error "Recording audio not possible permissions are missing"

nivas412, 2020-06-25:
# Bug Report

**To Do First**
```
       <RNCamera
                    ref={(ref) => {
                        this.camera = ref;
                    }}
                    style={styles.preview}
                    type={RNCamera.Constants.Type.back}
                    flashMode={RNCamera.Constants.FlashMode.on}
                    androidCameraPermissionOptions={{
                        title: 'Permission to use camera',
                        message: 'We need your permission to use your camera',
                        buttonPositive: 'Ok',
                        buttonNegative: 'Cancel',
                    }}
                    androidRecordAudioPermissionOptions={{
                        title: 'Permission to use audio recording',
                        message: 'We need your permission to use your audio',
                        buttonPositive: 'Ok',
                        buttonNegative: 'Cancel',
                    }}
                    onGoogleVisionBarcodesDetected={({ barcodes }) => {
                        console.log(barcodes);
                    }}
                    captureAudio={true}
                />
```

Faltform : Android

nivas412, 2020-08-01:
G:\Febbler2\febbler\android\app\src\main\AndroidManifest.xml

<uses-permission android:name="android.permission.RECORD_AUDIO" />

solves my issue

closed 2020-08-01 by nivas412

saczuac, 2021-04-14:
The issue comes when you record a video for the second time. RNCamera getting a warning "Recording audio not possible permissions are missing". Why ?