react-native-camera/react-native-camera #2881: Device/picture orientation are incorrect with specific landscape orientation

apalmer0, 2020-06-25:
# Bug Report

**To Do First**

- [X] Did you try latest release?
- [X] Did you try master?
- [X] Did you look for existing matching issues?

**Platforms**

iOS

**Versions**

- iOS: 13.5.1
- react-native-camera: 3.30.0
- react-native: 0.62.2
- react: 16.11.0
  

**Description/Current Behaviour**

When the phone is in landscape orientation and held perpendicular to the ground, both `pictureOrientation` and `deviceOrientation` in the `TakePictureResponse ` are correct (3, or "landscapeLeft"). While still in landscape orientation, if the phone is tilted forward about 45 degrees, those values are incorrect - they're both 1, or "portrait", and those photos don't get rotated correctly. 

image|TakePictureResponse|status
---|---|---
[image]|pictureOrientation: 3, deviceOrientation: 3|correct
[image]|pictureOrientation: 1, deviceOrientation: 1|incorrect

**Expected Behaviour**

Regardless (I think?) of the tilt of the phone, photos in landscape orientation should return a `pictureOrientation` and `deviceOrientation` of either 3 or 4, not 1, or at least the resulting image should be rotated to compensate for the picture having been taken in landscape orientation. That's the behavior of the native camera app at least.

**Steps to Reproduce**

Within the `takePicture` callback, log the `pictureOrientation` and `deviceOrientation` of `TakePictureResponse`. Turn the phone to the left for landscape orientation, and take a picture. Both orientations should be `3`, for `landscapeLeft`. Tilt the phone forward ~45+ degrees to take a picture of something on the floor. Both orientations will now be logged as `1`, which is "portrait", and incorrect.

```js
const App: () => React$Node = () => {
  const takePicture = async camera => {
    const data = await camera.takePictureAsync({
      fixOrientation: true,
      forceUpOrientation: true,
    });

    console.log('pictureOrientation', data.pictureOrientation);
    console.log('deviceOrientation', data.deviceOrientation);
    console.log('---------------------');
  };

  return (
    <>
      <StatusBar barStyle="dark-content" />
      <SafeAreaView>
        <ScrollView>
          <View>
            <RNCamera
              androidCameraPermissionOptions={androidCameraPermissionOptions}
              autoFocus="on"
              autoFocusPointOfInterest={{x: 0.5, y: 0.5}}
              captureAudio={false}
              flashMode="off"
              focusDepth={0}
              onBarCodeRead={null}
              onFaceDetected={null}
              onGoogleVisionBarcodesDetected={null}
              onTextRecognized={null}
              type="back"
              style={cameraStyles}
              useNativeZoom
              whiteBalance="auto"
              Xratio="16:9"
            >
              {({camera}) => (
                <TouchableOpacity onPress={() => takePicture(camera)}>
                  <Text> SNAP </Text>
                </TouchableOpacity>
              )}
            </RNCamera>
          </View>
        </ScrollView>
      </SafeAreaView>
    </>
  );
};
```

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

closed 2020-09-01 by stale[bot]

chriteixeira, 2020-09-15:
I'm experiencing the exact same problem. Is there an update or a workaround for this?

GuoMeng6, 2021-03-24:
Hi @chriteixeira @apalmer0 , I'm experiencing the exact same problem, how did you solved it. Thanks

apalmer0, 2021-03-24:
@GuoMeng6 I moved on without fixing it but it looks like @chriteixeira might have used a different library to get the device's orientation and then passed that orientation to the `orientation` option in the `TakePictureAsync` method? https://react-native-camera.github.io/react-native-camera/docs/api#takepictureasync

Just guessing here, seems like a good workaround if another library can more reliably track the device orientation

GuoMeng6, 2021-03-25:
@apalmer0 Thanks for your reply, i tried to use `react-native-orientation` and `react-native-orientation-locker` to listen device orientation but listener was not be called, i think my app using portrait mode. Anyway, thank you for help me.

chriteixeira, 2021-03-26:
Hey @GuoMeng6 . I did exactly as described by @apalmer0. I'm using react-native-orientation-locker to listen to device orientation changes and pass it to the camera options. Here's a small snippet:
```
const deviceAngle = useDeviceOrientation();
...
options.width =
  deviceAngle === 0 || deviceAngle === 180
    ? CameraPhoto.maxHeight
    : CameraPhoto.maxWidth;
    
await cameraRef.current.takePictureAsync(options);    
```

The _useDeviceOrientation()_ is a hook that will get the orientation and the picture width will be defined accordingly. Hope this helps.