react-native-camera/react-native-camera #2883: Camera crashing on Android emulator

richardfelkl, 2020-06-29:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [ ] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

Android Emulator

**Versions**

- Android: 9, 10
- react-native-camera: 3.30.0
- react-native: 0.62.2
- react: 16.11.0
  

**Description/Current Behaviour**

After upgrading to latest RN and RNCamera I started experiencing errors in Android emulator. I have tried two physical devides without problems. When calling takePictureAsync() app does nothing, no error shown on RN. Only Logcat throws bunch of errors regarding camera (Logcat output below). When I reload the page camera shows only black picture end throws RN error `Camera is already capturing`. Even the default camera app in system does not work.

**Expected Behaviour**

Camera will work without issues.

**Steps to Reproduce**

Use RNCamera as in example on Android emulator and call takePictureAsync().

**Logcat output**

```
2020-06-29 19:02:53.219 1620-4795/? V/EmulatedCamera_JPEGStub: JpegStub_compress: Compressed JPEG: 115200[320x240] -> 16949 bytes
2020-06-29 19:02:53.252 1620-4795/? V/EmulatedCamera_JPEGStub: JpegStub_compress: Compressed JPEG: 1843200[1280x960] -> 579631 bytes
    
    --------- beginning of crash
2020-06-29 19:02:53.253 1620-4795/? A/libc: Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0xf0e3b7ff in tid 4795 (EmulatedFakeCam), pid 1620 (provider@2.4-se)
2020-06-29 19:02:53.362 4798-4798/? I/crash_dump32: obtaining output fd from tombstoned, type: kDebuggerdTombstone
2020-06-29 19:02:53.365 1767-1767/? I//system/bin/tombstoned: received crash request for pid 4795
2020-06-29 19:02:53.370 4798-4798/? I/crash_dump32: performing dump of process 1620 (target tid = 4795)
2020-06-29 19:02:53.372 4798-4798/? A/DEBUG: *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
2020-06-29 19:02:53.372 4798-4798/? A/DEBUG: Build fingerprint: 'google/sdk_gphone_x86_64/generic_x86_64:9/PSR1.180720.120/6446634:userdebug/dev-keys'
2020-06-29 19:02:53.372 4798-4798/? A/DEBUG: Revision: '0'
2020-06-29 19:02:53.372 4798-4798/? A/DEBUG: ABI: 'x86'
2020-06-29 19:02:53.372 4798-4798/? A/DEBUG: pid: 1620, tid: 4795, name: EmulatedFakeCam  >>> /vendor/bin/hw/android.hardware.camera.provider@2.4-service <<<
2020-06-29 19:02:53.372 4798-4798/? A/DEBUG: signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0xf0e3b7ff
2020-06-29 19:02:53.372 4798-4798/? A/DEBUG:     eax ed100000  ebx ebf15f78  ecx 0008d82f  edx f0dae010
2020-06-29 19:02:53.372 4798-4798/? A/DEBUG:     edi f0dae010  esi f0dae010
2020-06-29 19:02:53.372 4798-4798/? A/DEBUG:     ebp ec06e788  esp ec06e768  eip f1073031
2020-06-29 19:02:53.381 4798-4798/? A/DEBUG: backtrace:
2020-06-29 19:02:53.382 4798-4798/? A/DEBUG:     #00 pc 0001b031  /system/lib/libc.so (memcpy+449)
2020-06-29 19:02:53.382 4798-4798/? A/DEBUG:     #01 pc 00001af8  /vendor/lib/hw/camera.goldfish.jpeg.so (JpegStub_getCompressedImage+56)
2020-06-29 19:02:53.382 4798-4798/? A/DEBUG:     #02 pc 0003e19b  /vendor/lib/hw/camera.ranchu.so (android::NV21JpegCompressor::getCompressedImage(void*)+59)
2020-06-29 19:02:53.382 4798-4798/? A/DEBUG:     #03 pc 0004b0f7  /vendor/lib/hw/camera.ranchu.so (android::JpegCompressor::compress()+535)
2020-06-29 19:02:53.382 4798-4798/? A/DEBUG:     #04 pc 0004b272  /vendor/lib/hw/camera.ranchu.so (android::JpegCompressor::threadLoop()+34)
2020-06-29 19:02:53.382 4798-4798/? A/DEBUG:     #05 pc 0000e656  /system/lib/vndk-sp-28/libutils.so (android::Thread::_threadLoop(void*)+326)
2020-06-29 19:02:53.382 4798-4798/? A/DEBUG:     #06 pc 0000dfb6  /system/lib/vndk-sp-28/libutils.so (thread_data_t::trampoline(thread_data_t const*)+246)
2020-06-29 19:02:53.382 4798-4798/? A/DEBUG:     #07 pc 0009cce5  /system/lib/libc.so (__pthread_start(void*)+53)
2020-06-29 19:02:53.382 4798-4798/? A/DEBUG:     #08 pc 00033c1b  /system/lib/libc.so (__start_thread+75)
2020-06-29 19:02:53.457 1902-2048/system_process W/NativeCrashListener: Couldn't find ProcessRecord for pid 1620
2020-06-29 19:02:53.462 1767-1767/? E//system/bin/tombstoned: Tombstone written to: /data/tombstones/tombstone_12
2020-06-29 19:02:53.472 1748-1849/? I/CameraProviderManager: Camera provider 'legacy/0' has died; removing it
2020-06-29 19:02:53.472 1748-1849/? I/CameraService: onDeviceStatusChanged: Status changed for cameraId=0, newStatus=0
2020-06-29 19:02:53.474 1748-1849/? I/CameraService: onDeviceStatusChanged: Client for camera ID 0 evicted due to device status change from HAL
2020-06-29 19:02:53.474 1748-1849/? E/Camera2Client: notifyError: Error condition 0 reported by HAL, requestId -1
2020-06-29 19:02:53.474 4046-4741/com.deliverytracking E/Camera: Error 2
2020-06-29 19:02:53.510 4805-4805/? I/android.hardware.camera.provider@2.4-service: Camera provider Service is starting.
[... 15 lines cut ...]
2020-06-29 19:02:53.532 4805-4805/? I/EmulatedCamera_FakeCamera3:   READ_SENSOR_SETTINGS
2020-06-29 19:02:53.532 4805-4805/? I/EmulatedCamera_FakeCamera3:   BURST_CAPTURE
2020-06-29 19:02:53.532 4805-4805/? I/EmulatedCamera_FakeCamera3:   FULL_LEVEL
2020-06-29 19:02:53.532 4805-4805/? I/EmulatedCamera_FakeCamera3:   (null)
2020-06-29 19:02:53.532 4805-4805/? E/EmulatedCamera_Factory: 2 cameras are being emulated. 1 of them are fake cameras.
2020-06-29 19:02:53.535 4805-4805/? E/EmulatedCamera_HotplugThread: createFileIfNotExists: Could not create file '/data/misc/media/emulator.camera.hotplug.0', error: 'Permission denied' (13)
2020-06-29 19:02:53.536 4805-4805/? E/EmulatedCamera_HotplugThread: createFileIfNotExists: Could not create file '/data/misc/media/emulator.camera.hotplug.1', error: 'Permission denied' (13)
2020-06-29 19:02:53.536 4805-4805/? I/CamProvider@2.4-impl: Loaded "Emulated Camera Module" camera module
2020-06-29 19:02:53.536 4805-4805/? E/CamProvider@2.4-impl: setUpVendorTags: Vendor tag operations not fully defined. Ignoring definitions.
2020-06-29 19:02:53.536 4805-4805/? E/CamProvider@2.4-impl: initialize: Vendor tag setup failed, will not be available.
2020-06-29 19:02:53.536 4805-4805/? V/EmulatedCamera_BaseCamera: getCameraInfo
2020-06-29 19:02:53.536 4805-4805/? V/EmulatedCamera_BaseCamera: getCameraInfo
2020-06-29 19:02:53.536 4805-4805/? I/ServiceManagement: Removing namespace from process name android.hardware.camera.provider@2.4-service to provider@2.4-service.
2020-06-29 19:02:53.541 4805-4805/? I/android.hardware.camera.provider@2.4-service: Registration complete for android.hardware.camera.provider@2.4::ICameraProvider/legacy/0.
2020-06-29 19:02:53.542 1748-1847/? I/CameraProviderManager: Connecting to new camera provider: legacy/0, isRemote? 1
2020-06-29 19:02:53.543 1748-1847/? I/CameraProviderManager: Enumerating new camera device: device@3.3/legacy/0
2020-06-29 19:02:53.545 1748-1847/? I/CameraProviderManager: Enumerating new camera device: device@3.3/legacy/1
2020-06-29 19:02:53.546 1748-1847/? I/CameraProviderManager: Camera provider legacy/0 ready with 2 camera devices
2020-06-29 19:02:53.547 1748-1847/? I/CameraService: onDeviceStatusChanged: Status changed for cameraId=1, newStatus=1
2020-06-29 19:02:53.547 1748-1847/? E/CameraService: onDeviceStatusChanged: State transition to the same status 0x1 not allowed
2020-06-29 19:02:53.547 1748-1847/? I/CameraService: onDeviceStatusChanged: Status changed for cameraId=0, newStatus=1
2020-06-29 19:02:56.325 1618-1712/? W/audio_hw_generic: Not supplying enough data to HAL, expected position 760741 , only wrote 601920
2020-06-29 19:02:56.478 1748-1849/? E/Camera3-Device: flush: Transaction error: Status(EX_TRANSACTION_FAILED): 'DEAD_OBJECT: '
2020-06-29 19:02:56.478 1748-1849/? E/Camera2Client: stopPreviewL: Camera 0: Unable to flush pending requests: Broken pipe (-32)
```

grkemtneri, 2020-07-03:
can you try on your real device ?

vovkasm, 2020-07-21:
We have very same problem. Camera works on device, but on emulator "Camera is already capturing" always returned by takePictureAsync

JustineAyroor, 2020-07-24:
Facing the same issue... using the bare workflow ... it works in a managed workflow

package.json:
"react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera.git",
"react-native": "~0.61.5",

Followed the instructions in the read me file for both IOS & Android
Works fine on IOS Emulator but on an Android Emulator when I click a picture (takePictureAsync) the whole app crashes.
Do we need any specific version of react-native package for this react-native-camera versions to work ??
Do we have make any more changes on the android/app/build.gradle file to make this work ???

winardis, 2020-08-06:
Hi me and my team are facing the same issue as well on sentry there are more than 2.4k recorded reports from our users, is there a way to resolve the issue?

package.json
"react-native": "0.62.2",
"react-native-camera": "^3.26.0",

[image]

meiqi1992, 2020-12-15:
Camera is already capturing same issue, resolve it? ,need help~ @richardfelkl

closed 2021-01-06 by stale[bot]

diguini, 2021-01-15:
Still Problem.

asugomez, 2022-03-09:
I have the same problem also