react-native-camera/react-native-camera #2888: Barcode Scanner not triggered on iOS

WrathChaos, 2020-07-06:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

Works great on **Android** but not even triggered on **iOS**

**Versions**

- iOS: 13.3.1
- react-native-camera: 3.31.0
- react-native: 0.61.5
- react: 16.12.0
- react-navigation:  4.0.10

**Description/Current Behaviour**

It does not scan the barcode or does not trigger the scanned method.

**Expected Behaviour**

It should scan the barcode without problem like Android.

## Reproducible Code:

```
<RNCamera
        ref={(ref) => {
          this.camera = ref;
        }}
        aspect={1}
        style={{ flex: 1 }}
        type={RNCamera.Constants.Type.back}
        onGoogleVisionBarcodesDetected={this.barcodeRecognized}
/ >

 barcodeRecognized = ({ barcodes }) => {
     alert("Barcodes: ", barcodes);
 }
```

tphuc012, 2020-07-08:
I have the same problem, please help me solve it

interstates21, 2020-07-10:
Same issue

vuongductuanktmt, 2020-07-15:
Same issue

vuongductuanktmt, 2020-07-15:
I used in Podfile pod 'react-native-camera', path: '../node_modules/react-native-camera', subspecs: [
  'TextDetector',
  'FaceDetectorMLKit',
  'BarcodeDetectorMLKit'
] it work

pearson-g, 2020-07-25:
@vuongductuanktmt  My problem is a bit different, IOS is triggered occasionally, I did not use CocoaPods, How do i configure thanks

closed 2020-08-11 by WrathChaos

ferasabufares, 2020-11-28:
pod 'react-native-camera', path: '../node_modules/react-native-camera', subspecs: [
'BarcodeDetectorMLKit'
]

can you help me please after add this to pod  when i try to pod install this error happened 
[image]

     react": "16.9.0",
    "react-native": "^0.63.3",
    "react-native-camera": "^3.40.0",

notjulian, 2020-12-01:
hi @ferasabufares,
In the same project I've also @react-native-firebase 

I add this on my Podfile

```
# Override Firebase SDK Version
$FirebaseSDKVersion = '6.34.0'
```

99uno, 2021-02-08:
Same issue: onGoogleVisionBarcodesDetected does not work on iOS, but I'm using react native 0.64, so can I add the line to the Podfile despite the react native version has autolinking?

stanica, 2021-11-26:
Is barcode detection dead on iOS?