react-native-camera/react-native-camera #2893: Is the version 3.31.0 compatible with RN 0.57 ?Also, where can I check the compatibility?

vishalenrique, 2020-07-09:
# Bug Report

Is the version 3.31.0 compatible with RN 0.57? Also, where can I check the compatibility?

Right now, when I'm trying that I'm getting the following errors: 

> /Users/vibhati/.gradle/caches/transforms-1/files-1.1/appcompat-v7-27.1.1.aar/7ce3c05cbebce5657b4b75995bfb6a95/res/values/values.xml:251:5-69: AAPT: error: resource android:attr/fontVariationSettings not found.
>     
> /Users/vibhati/.gradle/caches/transforms-1/files-1.1/appcompat-v7-27.1.1.aar/7ce3c05cbebce5657b4b75995bfb6a95/res/values/values.xml:251:5-69: AAPT: error: resource android:attr/ttcIndex not found.
>     
> error: failed linking references.

**Platforms**: Android

**Versions**

- react-native-camera:3.31.0
- react-native:0.57.8
- react:16.6.3
  

**Description/Current Behaviour**
Unable to build the project.

**Steps to Reproduce**

Integrate react-native-camera:3.31.0 with RN 0.57

sibelius, 2020-07-09:
I don't think this is a problem with react-native-camera

the latest breaking change was on v3 https://github.com/react-native-community/react-native-camera/releases/tag/v3.0.0

that you need to use react-native 60 because of AndroidX

sibelius, 2020-07-09:
feel free to send a PR improving the docs about the compatibility of RNCamera with react-native versions

closed 2020-07-09 by sibelius