react-native-camera/react-native-camera #2897: Manual Exposure too bright on iOS

ferbeab, 2020-07-16:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

iOS

**Versions**

- iOS: 13.6.0
- react-native-camera: 3.31.1
- react-native: 0.63.0
- react: 16.13.1
  -react-navigation: 5.6.2

**Description/Current Behaviour**

When manually defining exposure prop, minimum value (0) exposure is similar to autoExposure on normal light conditions. When exposure is increased over 0.2, the image gets super bright and saturated.

[image]

[image]

**Expected Behaviour**

Exposure in iOS should work like in Android. Image must look dark for values close to 0, normal around 0.5 and bright close to 1. 

**Steps to Reproduce**

Just run the following snippet in a iOS device. To check appearance for different exposure values, modify exposure prop with values from 0 to 1

```
 'use strict';
import React, { PureComponent } from 'react';
import { AppRegistry, StyleSheet, Text, TouchableOpacity, View } from 'react-native';
import { RNCamera } from 'react-native-camera';

class ExampleApp extends PureComponent {
  render() {
    return (
      <View style={styles.container}>
        <RNCamera
          ref={(ref) => {
            this.camera = ref;
          }}
          style={styles.preview}
          type={RNCamera.Constants.Type.back}
          flashMode={RNCamera.Constants.FlashMode.off}
          exposure={0.5}
          androidCameraPermissionOptions={{
            title: 'Permission to use camera',
            message: 'We need your permission to use your camera',
            buttonPositive: 'Ok',
            buttonNegative: 'Cancel',
          }}
          androidRecordAudioPermissionOptions={{
            title: 'Permission to use audio recording',
            message: 'We need your permission to use your audio',
            buttonPositive: 'Ok',
            buttonNegative: 'Cancel',
          }}
        />
        <View style={{ flex: 0, flexDirection: 'row', justifyContent: 'center' }}>
          <TouchableOpacity onPress={this.takePicture.bind(this)} style={styles.capture}>
            <Text style={{ fontSize: 14 }}> SNAP </Text>
          </TouchableOpacity>
        </View>
      </View>
    );
  }

  takePicture = async () => {
    if (this.camera) {
      const options = { quality: 0.5, base64: true };
      const data = await this.camera.takePictureAsync(options);
      console.log(data.uri);
    }
  };
}

const styles = StyleSheet.create({
  container: {
    flex: 1,
    flexDirection: 'column',
    backgroundColor: 'black',
  },
  preview: {
    flex: 1,
    justifyContent: 'flex-end',
    alignItems: 'center',
  },
  capture: {
    flex: 0,
    backgroundColor: '#fff',
    borderRadius: 5,
    padding: 15,
    paddingHorizontal: 20,
    alignSelf: 'center',
    margin: 20,
  },
});

AppRegistry.registerComponent('App', () => ExampleApp);
```

**Does it work with Expo Camera?**
No

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

closed 2020-09-22 by stale[bot]

gphloc, 2021-01-12:
I have same issue on IP 6, the exposure too bright when exposure={0.5} but when i change "aspect ratio" its work correct

vanhungoz, 2021-02-06:
@gpbl how did you change? could you provide a example?