react-native-camera/react-native-camera #2904: 3.31.0 leaks memory when scanning barcode

bozou, 2020-07-18:
# Bug Report

**To Do First**

- [No ] Did you try latest release?
- [Yes, but could neither run mlkit on device nor in simulator ] Did you try master?
- [Yes ] Did you look for existing matching issues?

**Platforms**

iOS

**Versions**

- Android: didn't try
- iOS:13.5.1
- react-native-camera:3.31.0
- react-native:0.63
- react:16.11.0
  
- react-navigation:5.5.0

**Description/Current Behaviour**
          &lt;RNCamera
            style={styles.preview}
            type={RNCamera.Constants.Type.back}
            onGoogleVisionBarcodesDetected={onGoogleBarCodeRead}
            zoom={0.007}
            captureAudio={false}
            >
Leave the barcode scanner camera on for several minutes, the app would crash (grow over 1GB).  I turned on react native Perf Monitor and can see memory increase quickly.

Reverted back to 3.27.1.  Memory no longer leaks.

**Expected Behaviour**
no memory leak.

**Steps to Reproduce**
see description.

**Additionals**

anhtuank7c, 2020-07-18:
I have the same issue.
My iPhone will run out of memory after 3-5 minutes then crash the app

closed 2020-10-04 by stale[bot]

marvintrajano, 2021-04-19:
Running into this issue as well, did anyone find a solution to this?