react-native-camera/react-native-camera #2905: onGoogleVisionBarcodesDetected output errorCode 3

code-by, 2020-07-18:
# Bug Report

**To Do First**

- [ x] Did you try latest release?
- [ x] Did you try master?
- [ x] Did you look for existing matching issues?

**Platforms**

**Versions**

- Android:
    "react": "16.13.1",
    "react-native": "0.63.1",
    "react-native-camera": "^3.32.0"  

**Description/Current Behaviour**

```
          onGoogleVisionBarcodesDetected={({ barcodes }) => {
            console.log(barcodes);
          }}
```
got in console
```
[{"bounds": {"origin": [Object], "size": [Object]}, "data": "{\"typeNumber\": \"0\", \"errorCode\": \"3\", \"version\": \"85\"}", "rawData": "{\"typeNumber\": \"0\", \"errorCode\": \"3\", \"version\": \"85\"}", "type": "UNKNOWN_FORMAT"}]
```

**Expected Behaviour**
I have printedout copy of loyalty cards with barcode type CODE_128.
Some barcode reader app from PlayStore could read it and display value very quickly

**Steps to Reproduce**
```
  onBarCodeRead = async (e) => {
    console.log(e);
    Alert.alert("Barcode value is " + e.data, "Barcode type is" + e.type);
  }

        <RNCamera
          captureAudio={false}
          style={styles.preview}
          onBarCodeRead={this.onBarCodeRead}
          ref={cam => this.camera = cam}
          onGoogleVisionBarcodesDetected={({ barcodes }) => {
            console.log(barcodes);
          }}
          googleVisionBarcodeMode={RNCamera.Constants.GoogleVisionBarcodeDetection.BarcodeMode.ALTERNATE}
        />
```

SainagChunduru, 2020-07-31:
Hi,

I am also facing the same issue. "onGoogleVisionBarcodesDetected" is always triggered even it is not focused on barcodes and returning error in data.

ssg-luke, 2020-08-05:
Yep seeing the same issue. Seems to have recently started without any code changes

wizebin, 2020-08-17:
Related to https://github.com/react-native-community/react-native-camera/issues/2875

closed 2021-01-13 by stale[bot]