react-native-camera/react-native-camera #2906: playSoundOnCapture false not working

kevinrichard31, 2020-07-18:
I seen many people where playSoundOnCapture false not working. im facing the same issue with french Android phone on react Native 0.62.1

I don't have shutter sound on instagram/fb or native camera app. Come from france, we don't have those restrictions.

So if someone could fix this issue...

thanks

cristianoccazinsp, 2020-07-21:
Let me know if you find a way to fix this. I have wondered this myself as well for a long time; the video API seems to always fire the shutter sound. Chances are it is by design, but I haven't had time to review this.

kevinrichard31, 2020-07-22:
@cristianoccazinsp found a way with usecamera2api={true}, but many people have crash or disfunctionnalities.
in my case while I change component and go back to camera component, the RNCamera view don't show the camera. and it look like laggy in comparison of the first Camera api version.

seirace, 2020-07-24:
Same problem for long time. I have many devices, the iOs ones are all working and some Android too but there is a shuttersound on some other android devices (Xiaomi mi9, OnePlus...) I didn't found a way to delete it except usecamera2api as @kevinrichard31 said but it make the others android devices crash (samsung galaxy s8)
Waiting for a solution too.

closed 2020-10-03 by stale[bot]