react-native-camera/react-native-camera #2925: iOS 14 Taking Picture

ayberkanilatsiz, 2020-07-24:
# Bug Report

**To Do First**

- [x] Did you try the latest release?
- [x] Did you try to master?
- [x] Did you look for existing matching issues?

**Platforms**

iOS

**Versions**

- iOS: **14.0**
- react-native-camera: master
- react-native: 0.61.5
- react: 16.9.0

**Description/Current Behaviour**

takePictureAsync function not working. It does not return anything. It stuck.

**Expected Behaviour**

taking picture

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

DevelopSmith, 2020-07-30:
@ayberkanilatsiz I have the same issue. Did you find a solution yet?

ayberkanilatsiz, 2020-07-30:
@DevelopSmith No, I don’t find any solution yet 😕

cavalos0086, 2020-08-23:
Hey guys @DevelopSmith  @ayberkanilatsiz  I am also learning about this tool but I have successfully used to take a picture so I would like to help if possible. 

Can you share some of your code?
Also have you enable the permissions in `info.plist`?

DevelopSmith, 2020-08-23:
@cavalos0086 I have tried it but didn't make any difference for me. I wasted many working hours trying to get this package to work but ended up using another package 😕

dkaushik94, 2020-09-18:
Has this been resolved? I get data back in an emulator but not in the device itself.

cavalos0086, 2020-09-19:
@dkaushik94 I haven't had any issues with this package but I could help. 
Do you have any snippet I can look at?

ayberkanilatsiz, 2020-09-25:
We changed the camera package we used and the problem was solved 👎 . I did not try again.

moahammadalt, 2020-10-12:
> @cavalos0086 I have tried it but didn't make any difference for me. I wasted many working hours trying to get this package to work but ended up using another package 😕

which another package have you used?

locomain, 2020-10-15:
Did anyone find a solution yet?

cavalos0086, 2020-10-15:
@locomain I got it working. What part doesn't work for you? Feel free to share some code

tushgup, 2020-10-29:
For me, takePictureAsync is working on iPad but not iPhone. Is anyone else facing this?

ShinsukeWatanabe, 2020-11-07:
@ayberkanilatsiz 
@DevelopSmith 
@dkaushik94 
@locomain 

This issue may occur on iPhone SE and iPhone SE 2 physical devices with iOS `14.0` and iOS `14.0.1`.

For our app, it works with the following versions:

react-native: `0.63.3`
react-native-camera: `3.40.0` or `2.11.2`
iOS: `14.1` or `14.2`

So we asked the user to update the iOS version.

williamsilveira45, 2020-11-19:
Same here on Iphone XR IOS 14.2

imamabdulazis, 2020-12-31:
Any solution please, 
My apps bug like this and customer was complain,

mousewip, 2021-01-11:
same issue on iPhone 11 with ios 14.0.1

ysfzrn, 2021-02-02:
same issue on iPhone 7 with ios 14.3

fesoares, 2021-02-12:
I don't know if it can help anyone, but I spent a lot of time trying to debug or change the plug-in, and restarting the iphone started working again

ayberkanilatsiz, 2021-03-17:
We tested again on iOS 14.4, it works fine.

- ( react-native-camera -> Master version )

I think this issue only appears on the beta version.

You should test on the release version.

ysfzrn, 2021-03-17:
Restarting phone resolved my issue but I replaced library with expo-camera

closed 2021-05-23 by stale[bot]