react-native-camera/react-native-camera #2928: Add burst / multiple capture (photo) support

aprilmintacpineda, 2020-07-28:
# Feature-Request

**Describe the Feature**

Currently we have `takePictureAsync ` which we call when we want to take a picture, and it takes one picture, but if you need to take `n` number of photos, say 10, the only approach I'm picturing is to call `takePictureAsync` succeedingly with `n` number of milliseconds interval. Something like so:

```js
const pictures = [];

function burstCapture () {
  const picture = await camera.takePictureAsync();
  pictures.push(picture);

 if (pictures.length < 10)  setTimeout(burstCapture, 100);
}

burstCapture();
```

**Possible Implementations**

Add option to `takePictureAsync`:

- `multiple<Boolean>` : indicates that this is a burst capture.
- `numCaptures` : The number of photos to be taken.
- `intervals`: The number of delays (milliseconds) between shots

**Related Issues**

Ignored: https://github.com/react-native-community/react-native-camera/issues/2133

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this feature to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

closed 2020-10-03 by stale[bot]

affantechcarrot, 2020-11-28:
@aprilmintacpineda have you found any solution for this functionality

aprilmintacpineda, 2020-11-28:
There's no built-in way to do this, I used [react-native-camera-kit](https://github.com/teslamotors/react-native-camera-kit) and then just called [capture](https://github.com/teslamotors/react-native-camera-kit#capture-----must-have-the-wanted-camera-capture-reference) several times in a row. react-native-camera-kit is faster when it comes to capture and that's why I used it instead even though it also doesn't have built-in support for burst capture.

Example:

```js
const takePicture = React.useCallback(async () => {
  // disable capture button while still taking picture
  setIsTakingPhoto(true);

  const images = await Promise.all(
    Array.from(Array(10), () => cameraRef.current.capture()),
  );

  // make sure to call save photo lastly so that it doesn't slow down the capturing
  await savePhotos(images);

  // done taking picture
  setIsTakingPhoto(false);
}, []);
```

farhanmeo, 2021-07-08:
Unable to create Burst Mode feature using React-native-camera