react-native-camera/react-native-camera #2934: Haptic Feedback doesn't work if `captureAudio` enabled

Jonovono, 2020-08-01:
Has anyone faced this? I am using the react native haptic feedback library. If I have the `captureAudio` prop on then haptic doesn't work, but if disabled its fine. I'm guessing it interferes but anyone know of a solution?

cristianoccazinsp, 2020-08-04:
Is it vibration that doesn't work?  I think I saw this once.  iOS treats vibration just like sound, so trying to play a vibration pattern while "sound is on" (i.e., `captureAudio=true`) will not work.  Your only option is to set it to `false` and only turn it on if you are in fact trying to record with sound.

nhannah, 2020-10-02:
> captureAudio=true

There are multiple threads about the limitation of haptic on iOS when the camera is in use. Haptic doesn't fire for a variety of reasons on iOS, low battery being another example, this is not a React Native issue but intended behavior on iOS.

closed 2021-01-13 by stale[bot]