react-native-camera/react-native-camera #2936: Taking a picture on Windows throws an error: "The application called an interface that was marshalled for a different thread."

shoaib, 2020-08-06:
# Bug Report

**To Do First**

- [ x ] Did you try latest release? Yes
- [ x ] Did you try master? Yes
- [ x ] Did you look for existing matching issues? Yes

**Platforms**
Windows

**Versions**

- Windows: 10
- react-native-camera: 3.35.0 (also tried git version)
- react-native: 0.62.0-0
- react: 16.11.0
  

**Description/Current Behaviour**

My app is throwing the following exception when trying to capture an image on Windows (Camera canvas is showing up fine on screen, but when I use takePictureAsync on the camera it seems to be breaking). I tried re-setting up the project multiple times with different version to no avail. Please note that the same code seems to be working perfectly on Android.

Possible Unhandled Promise Rejection (id: 0):
Error: HRESULT 0x8001010e
: The application called an interface that was marshalled for a different thread.
Error: HRESULT 0x8001010e
: The application called an interface that was marshalled for a different thread.
   at stop (Unknown script code:27435:11)
   at takePictureAsync$ (Unknown script code:118928:17)
   at tryCatch (Unknown script code:27041:9)
   at invoke (Unknown script code:27216:11)
   at Anonymous function (Unknown script code:27086:11)
   at tryCatch (Unknown script code:27041:9)
   at invoke (Unknown script code:27116:9)
   at Anonymous function (Unknown script code:27128:15)
   at tryCallOne (Unknown script code:29012:7)
   at Anonymous function (Unknown script code:29113:7)

**Expected Behaviour**

App should take the picture without any errors, or scan bar codes when a bar code is shown.

**Steps to Reproduce**

Set up react-native-windows project using the following instructions:
https://microsoft.github.io/react-native-windows/docs/getting-started

Follow react-native-camera windows installation instructions here:
https://react-native-community.github.io/react-native-camera/docs/installation

Set up following code in your render:
  render() {
    return (
      <View style={styles.container}>
        <RNCamera
          style={styles.preview}
          onBarCodeRead={this.onBarCodeRead.bind(this)}
          type={RNCamera.Constants.Type.front}
          flashMode={RNCamera.Constants.FlashMode.on}>
          {({camera, status}) => {
            if (status !== 'READY') {
              return <this.PendingView />;
            }
            return (
              <View
                style={{
                  flex: 0,
                  flexDirection: 'row',
                  justifyContent: 'center',
                }}>
                <TouchableOpacity
                  onPress={() => this.takePicture(camera)}
                  style={styles.capture}>
                  <Text style={{fontSize: 14}}> SNAP </Text>
                </TouchableOpacity>
              </View>
            );
          }}
        </RNCamera>
      </View>
    );
  }

Run the following code to capture the camera image:

  takePicture = async function (camera: RNCamera) {
    if (camera) {
      const options = {quality: 0.5, base64: true};
      const data = await camera.takePictureAsync(options);
      console.warn(data.uri);
    }
  };

Run the app on a Windows machine and hit "SNAP", it should throw the same error.

**Additionals**

Here's a screenshot of the exception:
[image]

Please let me know if you need any further information.

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

jonthysell, 2020-08-12:
Please update to 3.36.0, the issue was fixed in #2930.

shoaib, 2020-08-12:
> Please update to 3.36.0, the issue was fixed in #2930.

Thanks for this information, @jonthysell. I updated package version to 3.36.0, and the original issue seems to have resolved.
However, QR code scanning is still not working in Windows. It does work on Android, but not in Windows.
I tried using my own code, as well as replacing my entire app.js file with the code provided here:
"master/examples/basic/App.js"

There are no errors, no exceptions, camera and recording functionality are working fine, it's just the QR code scanning that's not working.

Can you please assist.

jonthysell, 2020-08-12:
@shoaib How are your running the app? We currently have a RNW bug (microsoft/react-native-windows#5639) when using `npx react-native run-windows` where the barcode binary doesn't get copied over when the app is deployed.

Current workaround is to build/run the app using F5 in Visual Studio for debug builds. FYI, `npx react-native run-windows --release` should still work from the command line and include the binaries.

shoaib, 2020-08-12:
> @shoaib How are your running the app? We currently have a RNW bug ([microsoft/react-native-windows#5639](https://github.com/microsoft/react-native-windows/issues/5639)) when using `npx react-native [...]

@jonthysell I am running the app using F5 on Visual Studio. Also tried using "npx react-native run-windows --release" but that didn't seem to help either. Any other reason you can think of why the QR code scanning wouldn't work?

shoaib, 2020-08-12:
All good. It's working as it should now. I cleaned the whole project, rebuilt the entire thing, and now it's working with the latest version of the package. Thanks for your help @jonthysell.

closed 2020-08-12 by shoaib