react-native-camera/react-native-camera #2937: rectOfInterest is not supported on onGoogleVisionBarcodesDetected with ML Kit

hamarles12, 2020-08-07:
# Bug Report

**To Do First**

- [x ] Did you try latest release?
- [x ] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

"react": "16.11.0",
"react-native": "0.62.2",
"react-native-camera": "^3.26.0"
"react-native-barcode-mask": "^1.2.2",

**Description/Current Behaviour**

As soon as a barcode enters the camera field, onGoogleVisionBarcodesDetected pick it up even if the barcode is on the outside of the finder area.

rectOfInterest works on onBarCodeRead but not on onGoogleVisionBarcodesDetected with ML Kit

**Expected Behaviour**

rectOfInterest should work on onGoogleVisionBarcodesDetected with ML Kit

GiannCarlo, 2020-08-07:
+1 .  Any solutions ?

LoverFancy, 2020-10-16:
same issues how to limit the face detection erea in react-native-camera

closed 2021-01-06 by stale[bot]

amirulasrof, 2021-05-05:
Same issue, rectOfInterest work fine onBarcodeRead but not onGoogleVisionBarcodesDetected.