react-native-camera/react-native-camera #2938: Android build not working when using react-native-camera and expo-camera together.

SmritiSYadav, 2020-08-07:
I want to use **react-native-camera** and **expo-camera** npms in my project for different modules. 

But since then in iOS everything is working fine. But in android I am not able to run the application. I am always getting following error.

I am been struggling to resolve this conflict. Any help will be really helpful.
I have raised this issue on [stack overflow](https://stackoverflow.com/questions/63180463/multiple-files-for-cameraview-while-build-app-for-android-react-native) as well but not able to get any help there as well.

```
error Failed to install the app. Make sure you have the Android development environment set up: https://reactnative.dev/docs/environment-setup. Run CLI with --verbose flag for more details.
Error: Command failed: ./gradlew app:installDebug -PreactNativeDevServerPort=8081

/Users/smritiibm/.gradle/caches/transforms-2/files-2.1/651d6b26e7e43a4bc4cbd7cbf706cbeb/jetified-cameraview-1.0.0-runtime.jar: D8: Type com.google.android.cameraview.Camera2$8 is defined multiple times: /Users/smritiibm/.gradle/caches/transforms-2/files-2.1/651d6b26e7e43a4bc4cbd7cbf706cbeb/jetified-cameraview-1.0.0-runtime.jar:com/google/android/cameraview/Camera2$8.class, /Users/smritiibm/Desktop/Projects/ReactNativeAccelerators/boilerplatestarterkit/node_modules/react-native-camera/android/build/intermediates/runtime_library_classes_jar/generalDebug/classes.jar:com/google/android/cameraview/Camera2$8.class
```

sibelius, 2020-08-07:
what is the motivation for this?

SmritiSYadav, 2020-08-07:
Hello @sibelius 

We are using `react-native-camera` for QR Code detection use case along with `react-native-qrcode-scanner` npm.

And we are using `expo-camera` for machine learning use case which is using `@tensorflow/tfjs-react-native npm`.

In standalone code both npms are working fine. 
But when I am trying to use both in my project, I am facing above issue for android build. iOS build is working properly without any issue.

sibelius, 2020-08-07:
maybe we can use a different namespace for cameraview to solve this issue

sibelius, 2020-08-07:
I think expo-camera has qrcode support

SmritiSYadav, 2020-08-09:
@sibelius 

Our project is in native cli. We are facing few issues implementing expo-camera. If you can provide any solution for this issue. It will be really helpful.

BrunoPes, 2020-08-10:
Hello folks!

I have the same issue: `react-native-camera` together with `expo-camera` is breaking on Android only - iOS works fine.
I'm able to build and open my app on Android, but it breaks only when I navigate to the specific screen using `react-native-camera` on **JSX**.

My usecase is the same as @SmritiSYadav, I have no option but to add `expo-camera` as it is one mandatory dependency for `@tensorflow/tfjs-react-native` (https://github.com/tensorflow/tfjs/tree/master/tfjs-react-native), even though the Camera component that I really want to display to the user is `react-native-camera`.

_Note: `RNCamera` was working perfectly on Android before I add `expo-camera`._

More details about my environment and error:
- "react": **16.11.0**
- "react-native": **0.62.2**
- "react-native-camera": **^3.31.0**
- "expo-camera": **^8.3.1**

**Android:**
- buildToolsVersion = **"29.0.3"**
- minSdkVersion = **21**
- compileSdkVersion = **29**
- targetSdkVersion = **28**
- **com.android.tools.build:gradle:3.5.2**
- gradle wrapper version = **6.0.1**

**Java error exception:**
```
08-10 12:27:02.799  8511  8511 E unknown:ViewManager: Error while updating prop playSoundOnCapture
08-10 12:27:02.799  8511  8511 E unknown:ViewManager: java.lang.reflect.InvocationTargetException
08-10 12:27:02.799  8511  8511 E unknown:ViewManager:   at java.lang.reflect.Method.invoke(Native Method)
08-10 12:27:02.799  8511  8511 E unknown:ViewManager:   at com.facebook.react.uimanager.ViewManagersPropertyCache$PropSetter.updateViewProp(ViewManagersPropertyCache.java:85)
08-10 12:27:02.799  8511  8511 E unknown:ViewManager:   at com.facebook.react.uimanager.ViewManagerPropertyUpdater$FallbackViewManagerSetter.setProperty(ViewManagerPropertyUpdater.java:136)
08-10 12:27:02.799  8511  8511 E unknown:ViewManager:   at com.facebook.react.uimanager.ViewManagerPropertyUpdater.updateProps(ViewManagerPropertyUpdater.java:56)
08-10 12:27:02.799  8511  8511 E unknown:ViewManager:   at com.facebook.react.uimanager.ViewManager.updateProperties(ViewManager.java:49)
08-10 12:27:02.799  8511  8511 E unknown:ViewManager:   at com.facebook.react.uimanager.NativeViewHierarchyManager.createView(NativeViewHierarchyManager.java:279)
08-10 12:27:02.799  8511  8511 E unknown:ViewManager:   at com.facebook.react.uimanager.UIViewOperationQueue$CreateViewOperation.execute(UIViewOperationQueue.java:180)
08-10 12:27:02.799  8511  8511 E unknown:ViewManager:   at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.dispatchPendingNonBatchedOperations(UIViewOperationQueue.java:987)
08-10 12:27:02.799  8511  8511 E unknown:ViewManager:   at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded(UIViewOperationQueue.java:958)
08-10 12:27:02.799  8511  8511 E unknown:ViewManager:   at com.facebook.react.uimanager.GuardedFrameCallback.doFrame(GuardedFrameCallback.java:29)
08-10 12:27:02.799  8511  8511 E unknown:ViewManager:   at com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher.doFrame(ReactChoreographer.java:175)
08-10 12:27:02.799  8511  8511 E unknown:ViewManager:   at com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1.doFrame(ChoreographerCompat.java:85)
08-10 12:27:02.799  8511  8511 E unknown:ViewManager:   at android.view.Choreographer$CallbackRecord.run(Choreographer.java:964)
08-10 12:27:02.799  8511  8511 E unknown:ViewManager:   at android.view.Choreographer.doCallbacks(Choreographer.java:790)
08-10 12:27:02.799  8511  8511 E unknown:ViewManager:   at android.view.Choreographer.doFrame(Choreographer.java:721)
08-10 12:27:02.799  8511  8511 E unknown:ViewManager:   at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:951)
08-10 12:27:02.799  8511  8511 E unknown:ViewManager:   at android.os.Handler.handleCallback(Handler.java:883)
08-10 12:27:02.799  8511  8511 E unknown:ViewManager:   at android.os.Handler.dispatchMessage(Handler.java:100)
08-10 12:27:02.799  8511  8511 E unknown:ViewManager:   at android.os.Looper.loop(Looper.java:214)
08-10 12:27:02.799  8511  8511 E unknown:ViewManager:   at android.app.ActivityThread.main(ActivityThread.java:7356)
08-10 12:27:02.799  8511  8511 E unknown:ViewManager:   at java.lang.reflect.Method.invoke(Native Method)
08-10 12:27:02.799  8511  8511 E unknown:ViewManager:   at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
08-10 12:27:02.799  8511  8511 E unknown:ViewManager:   at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:930)
08-10 12:27:02.799  8511  8511 E unknown:ViewManager: Caused by: java.lang.NoSuchMethodError: No virtual method setPlaySoundOnCapture(Z)V in class Lorg/reactnative/camera/RNCameraView; or its super classes (declaration of 'org.reactnative.camera.RNCameraView' appears in /data/app/com.thefarmproject.lettucegrow.staging-4N5twDpR3TU1NGWHZ3bfDA==/base.apk!classes38.dex)
08-10 12:27:02.799  8511  8511 E unknown:ViewManager:   at org.reactnative.camera.CameraViewManager.setPlaySoundOnCapture(CameraViewManager.java:140)
08-10 12:27:02.799  8511  8511 E unknown:ViewManager:   ... 23 more
08-10 12:27:02.816  8511  8511 E unknown:ReactNative: Exception in native call
08-10 12:27:02.816  8511  8511 E unknown:ReactNative: com.facebook.react.bridge.JSApplicationIllegalArgumentException: Error while updating property 'playSoundOnCapture' of a view managed by: RNCamera
[... cut ...]
08-10 12:27:02.833  8511  8511 E AndroidRuntime:        at java.lang.reflect.Method.invoke(Native Method)
08-10 12:27:02.833  8511  8511 E AndroidRuntime:        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
08-10 12:27:02.833  8511  8511 E AndroidRuntime:        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:930)
08-10 12:27:02.835  1984  2086 W ActivityTaskManager:   Force finishing activity com.thefarmproject.lettucegrow.staging/com.lettucegrow.MainActivity
```

dondragon2, 2020-11-03:
@BrunoPes did you find a solution to this issue?

phantom0216, 2021-04-13:
face the same problem

znoland3, 2021-04-30:
facing same problem

eliottguillaumin, 2021-05-17:
Same problem, I'm now struggling to remove expo-camera, how can I even do this?
Removing an expo module in a bare workflow app isn't documented and last I tried npm uninstall --save expo-camera it didn't fully do the job and the app wouldn't even compile.

closed 2021-07-23 by stale[bot]

AKACC, 2021-08-19:
> Same problem, I'm now struggling to remove expo-camera, how can I even do this?
> Removing an expo module in a bare workflow app isn't documented and last I tried npm uninstall --save expo-camera it didn't fully do the job and the app wouldn't even compile.
> 
> EDIT: Had to manually disable the expo-camera unimodule in the android gradles and podfile then gradlew clean
> There is no straightforward way to uninstall expo modules, uninstallation of packages should be implemented in docs

How did you manually disable expo-camera? Could you share the steps?

SpectreWulf, 2022-08-23:
> EDIT: Had to manually disable the expo-camera unimodule in the android gradles and podfile then gradlew clean There is no straightforward way to uninstall expo modules, uninstallation of packages should be implemented in docs

Hi @eliottguillaumin @AKACC 

Facing the same issue in an expo bare workflow!
Can you please list the steps to disable the expo-camera?

nthiendev, 2022-10-24:
<p style="box-sizing: border-box; margin-top: 0px; margin-bottom: 20px; color: rgb(17, 17, 17); font-size: 18px; line-height: 3rem !important; font-family: Roboto, &quot;Helvetica Neue&quot;, Helvetica, Arial, sans-serif; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: left; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">nguyên Bí thư Huyện uỷ, nguyên Chủ tịch HĐND H.Cam Lâm, tỉnh Khánh Hoà; Hà Quang Dĩnh, nguyên Tỉnh ủy viên, nguyên Bí thư Ban Cán sự Đảng, nguyên Chánh án TAND tỉnh Hoà Bình.</p>

 
--
Tại kỳ họp thứ 21, Ủy ban Kiểm tra T.Ư đã quyết định đề nghị Bộ Chính trị kỷ luật Ban Cán sự Đảng Bộ GD-ĐT nhiệm kỳ 2016 - 2021 và ông Phùng Xuân Nhạ, Phó trưởng Ban Tuyên giáo T.Ư, nguyên Ủy viên T.Ư Đảng, nguyên Bí thư Ban cán sự đảng, nguyên Bộ trưởng Bộ GD-ĐT.