react-native-camera/react-native-camera #2949: unable to build for release on Android

ahdam, 2020-08-17:
# Question

**To Do First**
- [x] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [x] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [x] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Versions**

`"react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera.git",`
npm list react-native-camera -> `└── react-native-camera@3.36.0  invalid`

- react-native-qrcode-scanner: 1.4.1,
- react-native-camera: 3.36.0
- react-native: 0.60.5
- react: 16.8.6

I'm using `react-native-camera` with `react-native-qrcode-scanner`, so the feature I need is QR Scanning so I followed all the steps from the documentation except the additional ones for Face Detection/Text Recognition/BarCode with MLKit (because I read in another question that I don't need those steps). 

- iOS dev build: OK
- iOS release build: OK
- Android debug build: OK
- Android release build: FAIL

```
FAILURE: Build failed with an exception.
* What went wrong:

Could not determine the dependencies of task ':react-native-camera:compileMlkitReleaseJavaWithJavac'.
> Could not resolve all task dependencies for configuration ':react-native-camera:mlkitReleaseCompileClasspath'.
   > Could not find com.google.android.gms:play-services-vision-image-label:11.0.0.
     Required by:
         project :react-native-camera > com.google.firebase:firebase-ml-vision:19.0.3
   > Could not find com.google.android.gms:play-services-flags:11.0.0.
     Required by:
         project :react-native-camera > com.google.firebase:firebase-ml-vision-face-model:17.0.2
   > Could not find com.google.android.gms:play-services-phenotype:11.0.0.
     Required by:
         project :react-native-camera > com.google.firebase:firebase-ml-common:17.0.0
   > Could not find com.google.android.gms:play-services-stats:11.0.0.
     Required by:
         project :react-native-camera > com.google.firebase:firebase-iid:17.0.3
```

**app/build.graddle**
```
apply plugin: "com.android.application"
...
project.ext.react = [
    entryFile: "index.js",
    enableHermes: false,  // clean and rebuild if changing
]

apply from: "../../node_modules/react-native/react.gradle"
def enableSeparateBuildPerCPUArchitecture = false
def jscFlavor = 'org.webkit:android-jsc-intl:+'
ableHermes = project.ext.react.get("enableHermes", false);
def enableProguardInReleaseBuilds = false

android {
    compileSdkVersion 23
    buildToolsVersion rootProject.ext.buildToolsVersion

    compileOptions {
         sourceCompatibility JavaVersion.VERSION_1_8
         targetCompatibility JavaVersion.VERSION_1_8
     }

    defaultConfig {
        applicationId "com.myapp"
        minSdkVersion rootProject.ext.minSdkVersion
        targetSdkVersion rootProject.ext.targetSdkVersion
        versionCode 78
        versionName "2.1.7"
        missingDimensionStrategy 'react-native-camera', 'general'
       ...
    }
}

dependencies {
  ...  
  implementation project(':react-native-camera')
  ...
}
...
apply from: file("../../node_modules/@react-native-community/cli-platform-android/native_modules.gradle"); applyNativeModulesAppBuildGradle(project)

```

**build.graddle**
```
buildscript {
    repositories {
        jcenter()
        maven {
            url 'https://maven.google.com/'
            name 'Google'
        }
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:3.3.0'
    }
}
subprojects {
    project.configurations.all {
        resolutionStrategy.eachDependency { details ->
            if (details.requested.group == 'com.android.support'
                    && !details.requested.name.contains('multidex') ) {
                details.useVersion "26.1.0"
            }
        }
    }

    afterEvaluate {project ->
        if (project.hasProperty("android")) {
            android {
                compileSdkVersion 28     //do this in android/app/build.gradle too
                buildToolsVersion = "28.0.3"  //do this in android/app/build.gradle too
            }
        }
    }

}

allprojects {
    repositories {
        configurations.all {
        resolutionStrategy.eachDependency { DependencyResolveDetails details ->
        def requested = details.requested
           if (requested.group == 'com.google.android.gms') {
              details.useVersion '11.0.0'
           }
           }
         }
        maven {
                    url 'https://maven.google.com/'
                    name 'Google'
                }
        jcenter()
        maven {
            url "$rootDir/../node_modules/react-native/android"
        }
        maven {
            // Android JSC is installed from npm
            url("$rootDir/../node_modules/jsc-android/dist")
        }

        maven { url "https://jitpack.io" }
    }
}

ext {
    buildToolsVersion = "28.0.3"
    minSdkVersion = 16
    compileSdkVersion = 28
    targetSdkVersion = 28
    supportLibVersion = "28.0.0"
    googlePlayServicesVersion = "16.0.0"
    androidMapsUtilsVersion = "0.5+"
}
```

**MainApplicaton.java**
```
package myapp;
import android.app.Application;
import android.util.Log;
import org.reactnative.camera.RNCameraPackage;
import com.reactnativecommunity.rnpermissions.RNPermissionsPackage;

import java.util.Arrays;
import java.util.List;

public class MainApplication extends Application implements ReactApplication {
  private static CallbackManager mCallbackManager = CallbackManager.Factory.create();
  protected static CallbackManager getCallbackManager() {
    return mCallbackManager;
  }
  private final ReactNativeHost mReactNativeHost = new ReactNativeHost(this) {
    @Override
    public boolean getUseDeveloperSupport() {
      return BuildConfig.DEBUG;
    }
    @Override
    protected List<ReactPackage> getPackages() {
      return Arrays.<ReactPackage>asList(
        new MainReactPackage(),
        new RNPermissionsPackage(),
        new RNCameraPackage(),
        ...
      );
    }
    @Override
    protected String getJSMainModuleName() {
      return "index";
    }
  };
  @Override
  public ReactNativeHost getReactNativeHost() {
    return mReactNativeHost;
  }
  @Override
  public void onCreate() {
    super.onCreate();
    AppEventsLogger.activateApp(this);
    SoLoader.init(this, /* native exopackage */ false);
    Reprint.initialize(this);
  }
}
```

**gradle-wrapper.properties**
```
...
distributionUrl=https\://services.gradle.org/distributions/gradle-5.4.1-all.zip
...
```

I've been wracking my brains on this one for some time now. Any help is highly appreciated. Thanks.

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

ahdam, 2020-08-17:
I figured it out:

1. the autolink feature doesn't work for this module, so manual installation is required
2. **build.graddle**

```
...
ext {
    ...
    googlePlayServicesVersion = "16.0.0"
}
...
```

This is what worked for me, for the setup described above. 
Hope this saves someone else's time 😄 🖖

closed 2020-08-17 by ahdam

dragma, 2021-02-16:
> I figured it out:
> 
> 1. the autolink feature doesn't work for this module, so manual installation is required
> 2. **build.graddle**
> 
> ```
> ...
> ext {
>     ...
>     googlePlayServicesVisionVersion = "19.0.0" <-- add this
> }
> ...
> ```
> 
> This is what worked for me, for the setup described above.
> Hope this saves someone else's time

Wow. Thanks a lot ! This saved me !