react-native-camera/react-native-camera #2951: Possibility to take picture in RAW format ?

mploux, 2020-08-19:
Hello,

I was wondering how to capture RAW images, I wonder if it is even possible to do it with RNCamera. 
I know for a fact that it's possible to capture RAW images on a lot of phones, and a lot of application allow us to do so.
But I didn't find anything on google or in the docs on how to do it. Do I have to code everything from scratch or is there a way to do it using RNCamera ?
What does 'quality: 1' qive us exactly ? Is it just a jpeg without any compression ? (Still not raw) Or is it real Raw ?
Do I have to use base64 ? Obv I need to disable 'skipProcessing'
I find it hard to believe that I'm de only one looking to get RAW images.
I saw this issue which was closed for inactivity #2675 

Thanks for this great package !

qpre, 2020-09-07:
You're not the only one !

It'd be nice if we could get the original data and/or associated RAW data as the result of taking a picture (thus allowing the developer to process in other ways than the packaged ones).

I'm not familiar enough with the code base to take it on right now but does anyone know what it would take ?

mploux, 2021-01-07:
@qpre Apparently we are...

HaydenPalmer, 2021-02-12:
Any updates on this? Would be great if we could access the RAW camera. I was looking into the possibility of incorporating this: https://opencamera.org.uk/

MarcoScabbiolo, 2021-02-12:
I don't think it is viable to get the raw byte array in the JS side and processing that byte array on the JS side would be painfully slow.

A better solution would be to facilitate getting the raw byte array in the native side. This kind of extensibility is something we're working on [v4](#3100)

umerDev, 2021-04-21:
any updates on this, would be useful if we could capture images in dng format.

closed 2021-06-27 by stale[bot]