react-native-camera/react-native-camera #2957: CameraManager.checkVideoAuthorizationStatus is not a function on IOS

Gerrist, 2020-08-26:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

IOS

**Description/Current Behaviour**

I successfully manually installed everything for IOS, but I still get the 'CameraManager.checkVideoAuthorizationStatus is not a function' error.

**Expected Behaviour**

Unknown

**Steps to Reproduce**

<RNCamera
                onBarCodeScanned={(cb) => {
                    console.log(cb);
                }}
                captureAudio={false}
                // style={StyleSheet.absoluteFillObject}
                style={{
                    width: Dimensions.get('window').width,
                    height: Dimensions.get('window').width,
                }}
            />

liqili, 2020-09-22:
I got this error when it was running on top of expo client.

sniperosx, 2020-09-23:
@Gerrist did you find a solution to this problem?
Having this problem after moving from Expo to a vanilla react native app. 
(Replacing the expo components)

Did i forget to remove some legacy expo code?

Pod is up to date, everything cleaned and run on device (not simulator). 
Checked the docs more than once, but don't get this part running.

Gerrist, 2020-10-08:
> @Gerrist did you find a solution to this problem?

Unfortunately not. Still looking for a solution

sniperosx, 2020-10-08:
Okay @Gerrist same here. 
We're dropped the framework completeley and implemented the feature we needed as native modules ;)

closed 2021-01-13 by stale[bot]

moigamijunior, 2022-01-25:
Same issue! Some news?

x122536638, 2022-03-16:
if not  pod install  will show this errot: Possible Unhandled Promise Rejection (id: 1):
TypeError: CameraManager.checkVideoAuthorizationStatus