react-native-camera/react-native-camera #2959: Could not resolve com.google.android.gms:play-services-vision-common:[17.0.2].

wutiange, 2020-08-31:
上一次还能正常运行，但是在今天我重新跑项目的时候突然就出现了这个错误，我按照网上的方法，也包括本插件的issure，发现并不能解决这个，同时我也发现只要是跟 com.google.android.gms 相关的都会错误
Last time, it was able to run normally, but this error suddenly occurred when I ran the project again today(2020-08-31). According to the methods on the Internet, including the issue of this react-native-camera, I found that it could not solve this problem. At the same time, I also found that as long as I was following the com.google.android.gms related errors

closed 2020-08-31 by wutiange

zhaiduo, 2020-11-15:
其实就是强的问题

wutiange, 2020-11-16:
感谢，嗯，我也发现了，我换成流量就OK了

nolightstar, 2020-12-17:
兄弟，所以最后是怎么操作的，开热点打包么

wutiange, 2020-12-18:
> 兄弟，所以最后是怎么操作的，开热点打包么

如果公司的网跑出现这个问题，使用热点就可以了，我是这样解决的

nolightstar, 2020-12-18:
好的非常感谢

IT-LiuSir, 2021-01-12:
> 其实就是强的问题

不是墙的问题呢，移动联通，热点都试过，就是无法解决

IT-LiuSir, 2021-01-12:
> 上一次还能正常运行，但是在今天我重新跑项目的时候突然就出现了这个错误，我按照网上的方法，也包括本插件的issure，发现并不能解决这个，同时我也发现只要是跟com .google.android.gms相关的都会错误
> 上一次，它能够正常运行，但是当我今天（2020-08-31）再次运行该项目时，此错误突然发生。根据互联网上的方法，包括此“本机相机”的问题，我发现它无法解决此问题。同时，我还发现，只要遵循com.google.android.gms相关错误

请问如何解决的

wutiange, 2021-01-15:
> > 上一次还能正常运行，但是在今天我重新跑项目的时候突然就出现了这个错误，我按照网上的方法，也包括本插件的issure，发现并不能解决这个，同时我也发现只要是跟com .google.android.gms相关的都会错误
> > 上一次，它能够正常运行，但是当我今天（2020-08-31）再次运行该项目时，此错误突然发生。根据互联网上的方法，包括此“本机相机”的问题，我发现它无法解决此问题。同时，我还发现，只要遵循com.google.android.gms相关错误
> 
> 请问如何解决的

肯定是跟网有关。如果实在不行，你可以翻墙试试。