react-native-camera/react-native-camera #2966: Face bounding box is consistently off for android devices

sbycrosz, 2020-09-07:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: Android 10
- iOS: N/A
- react-native-camera: 3.36.0
- react-native: 61.5
- react:16.9.0

**Description/Current Behaviour**

[image]

Face bounding box is consistently off for android devices

Meanwhile, iOS is working as expected.

**Expected Behaviour**

Face bounding box should bound faces 

**Steps to Reproduce**

```
<View style={[s.absolute_fill]}>
  <RNCamera
    ref={camera}
    type='front'

    autoFocus={RNCamera.Constants.AutoFocus.on}
    captureAudio={false}

    androidCameraPermissionOptions={{
      title: t('self-portrait.viewfinder.android-permission-dialog.title'),
      message: t('self-portrait.viewfinder.android-permission-dialog.message'),
      buttonPositive: t('self-portrait.viewfinder.android-permission-dialog.button-positive'),
      buttonNegative: t('self-portrait.viewfinder.android-permission-dialog.button-negative'),
    }}

    onFacesDetected={onFacesDetected}

    style={[s.absolute_fill]} />
    
  {renderFaces()}
</View>
```

cristianoccazinsp, 2020-09-08:
Does it work with landscape?

sbycrosz, 2020-09-08:
Hi, 

We found out that setting a ratio on RNCamera will fix the bounding box issues.

Then, we try using getSupportedRatiosAsync to find the best match ratio given device resolution. 
But we found that changing the ratio prop when RNCamera is mounted will somehow stretch the preview image horizontally

I think we are close.

Cheers,

sbycrosz, 2020-09-08:
Hi, 

With further testing, we found that:

1. On landscape, faces seem to be rotated by 90 degrees
2. Setting up the ratio prop before RNCamera is rendered would fix this issue (faces detected correctly)
3. We can't do (2) because getSupportedRatioAsync can only be called after RNCamera is rendered
4. Changing the ratio prop after RNCamera is rendered would cause the preview image to be stretched horizontally

Cheers, 
Sam

(1)
[image]

(4)
[image]

cristianoccazinsp, 2020-09-08:
Were you trying 4:3 and 16:9 ratios? I think you may have an issue with the camera taking the whole screen (which is technically incorrect, look at the native camera). 

For example, if you are using 4:3, you should build your view dimensions so it fits into a 4:3 rectangle. The following picture shows a 4:3 view that takes as much height/width (based on on portrait or landscape) as possible while leaving the other one adjust automatically.

[image]

So if you are using 4:3, you will need the black lines on the sides (I think). If you are using 16:9, then you may be able to take the full dumensions of the screen on some devices.

sbycrosz, 2020-09-09:
[image]

Interesting. so I guess we could probably set the RNCamera ratio to 4:3 & render it slightly bigger than the screen for a workaround 

And even more interesting, we found this [aspect prop](https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/lwansbrough/RCTCamera/RCTCameraView.java#L195) that seems to do that (but we haven't manage to get it working yet)

closed 2021-01-13 by stale[bot]