react-native-camera/react-native-camera #2971: option playSoundOnCapture={false} not work in the determined devices

joseviniciusnunes, 2020-09-09:
Hello guys, just a doubt the option playSoundOnCapture = {false} does not work on certain devices, example Xiaomi MI A3 the sound is always triggered, on AVD and LG devices it did not occur to me, do you know if it is possible to get around?

Thank you for the excellent work!

D4nt4lion, 2020-09-23:
On some Android phones I needed to turn off the shutter sound function in the camera menu on the phone itself:
https://www.youtube.com/watch?time_continue=47&v=qhpBodXIDGA&feature=emb_logo

I think some phones force the Camera Sound and have no option for turning it off. More so certain Countries (f.e. Japan) do not allow to turn off the shutter sound.

I am not sure about why it is like this though.

jladuval, 2020-11-16:
I'm also experiencing this issue on a Pixel 3. All settings have it turned off but it still plays the shutter sound when capturing video.

closed 2021-01-22 by stale[bot]