react-native-camera/react-native-camera #2974: Error when pod install FirebaseML kit

Nadreasky, 2020-09-16:
I have followed instruction guide when install react-native-camera library for my project. Because I want to use Barcode and Text Recognition so I have modified my Podfile by adding below code

`pod 'react-native-camera', path: '../node_modules/react-native-camera', subspecs: [
  'TextDetector',
  'BarcodeDetectorMLKit'
  ]`

but when run command `pod install`, I got a warning like:

[!] FirebaseMLVisionBarcodeModel has been deprecated in favor of MLKitBarcodeScanning

[!] FirebaseMLVisionTextModel has been deprecated in favor of MLKitTextRecognition

Not sure my app is still working or not (because I'm still writing code for now), but I think sooner or later I need to update those deprecated pod.

Please help me to check and give me advices for this case. Thank you very much!

BooYeu, 2020-10-23:
I have same problem

ramezyouakimf, 2020-11-02:
I have the same problem

sarojswain680, 2020-11-05:
I have the same problem

Donhv, 2020-11-12:
Does anyone maintain this repo?

closed 2021-01-18 by stale[bot]

GiannCarlo, 2021-02-08:
still an issue

faizantariq1, 2021-02-17:
still got the warning. [!] FirebaseMLVisionTextModel has been deprecated in favor of MLKitTextRecognition. when will you guys convert it into MLKitTextRecognition?

KPS250, 2021-06-17:
Still facing the same issue. QRCode Scanning is working for Android but not for iOS. Any update  on this?

notjulian, 2021-06-17:
Hi
I've patched the latest version this way:

```
diff --git a/node_modules/react-native-camera/react-native-camera.podspec b/node_modules/react-native-camera/react-native-camera.podspec
index a67d1fd..3df9191 100644
--- a/node_modules/react-native-camera/react-native-camera.podspec
+++ b/node_modules/react-native-camera/react-native-camera.podspec
@@ -26,22 +26,19 @@ Pod::Spec.new do |s|
   s.subspec "TextDetector" do |ss|
     ss.dependency 'react-native-camera/RN'
     ss.dependency 'react-native-camera/RCT'
-    ss.dependency 'Firebase/MLVision'
-    ss.dependency 'Firebase/MLVisionTextModel'
+    ss.dependency 'GoogleMLKit/BarcodeScanning'
   end
 
   s.subspec "FaceDetectorMLKit" do |ss|
     ss.dependency 'react-native-camera/RN'
     ss.dependency 'react-native-camera/RCT'
-    ss.dependency 'Firebase/MLVision'
-    ss.dependency 'Firebase/MLVisionFaceModel'
+    ss.dependency 'GoogleMLKit/BarcodeScanning'
   end
 
   s.subspec "BarcodeDetectorMLKit" do |ss|
     ss.dependency 'react-native-camera/RN'
     ss.dependency 'react-native-camera/RCT'
-    ss.dependency 'Firebase/MLVision'
-    ss.dependency 'Firebase/MLVisionBarcodeModel'
+    ss.dependency 'GoogleMLKit/BarcodeScanning'
   end
 
   s.default_subspecs = "RN", "RCT"
```

```
"react-native": "0.64.2"
"react-native-camera": "^3.43.9"
```

KPS250, 2021-06-17:
@notjulian thanks for the quick help. I was able to successfully install the pods and run the project but however its not detecting QRCodes/BarCodes  on iOS.

notjulian, 2021-06-17:
> @notjulian thanks for the quick help. I was able to successfully install the pods and run the project but however its not detecting QRCodes/BarCodes on iOS.

are you using this method?

`onBarCodeRead={this.onBarCodeRead}`

wojteg1337, 2021-06-22:
@notjulian you should specify bar code types

KPS250, 2021-06-22:
> > @notjulian thanks for the quick help. I was able to successfully install the pods and run the project but however its not detecting QRCodes/BarCodes on iOS.
> 
> I'm using this code ... ( please check `onBarCodeRead`)
> 
> ```
> <RNCamera
>       style={styles.preview}
>       type={RNCamera.Constants.Type.back}
>       captureAudio={false}
>       //flashMode={RNCamera.Constants.FlashMode.on}
>       onBarCodeRead={this.onBarCodeRead}></RNCamera>
> ```

This worked!

mahecode, 2023-04-10:
onTextRecognized but this is not working any help here ?