react-native-camera/react-native-camera #2978: incorrect scanned code

gersualves775, 2020-09-17:
# Bug Report

- [ X] Did you try latest release?
- [ X] Did you try master?
- [ X] Did you look for existing matching issues?

- Android:
- iOS:
- react-native-camera: 3.29.1 / 3.23.1
- react-native: 0.62.1
- react: 16.11.0

[image]

Hello, my problem is with barcode reading
this for example, your code is
23793 38128 60039 508019 87000 063302 1 83820000002000

but the native reactor-camera insists on returning me another code.

gersualves775, 2020-10-07:
up

Ashafix, 2020-10-09:
Strange, I tried two other programs (Android app: QR & Barcode Scanner and https://online-barcode-reader.inliteresearch.com/). Both return the same text: 00192832500000096430000003197150014510642017.

Is it possible that the software which generated the barcode misinterpreted your input?

MateusAndrade, 2020-10-09:
This is Brazillian Boleto @gersualves775 ?

gersualves775, 2020-10-14:
Yes @MateusAndrade.
The problem is that I always used it and it worked. After updating some features, the problem occurs.

gersualves775, 2020-10-14:
@Ashafix 
me too.
But the RNCamera return  23793 38128 60039 508019 87000 063302 1 83820000002000

Ashafix, 2020-10-14:
@gersualves775 
I tried with RNCamera 3.36 and React Native 0.62. The code was correctly recognized in an emulated device.

closed 2021-01-13 by stale[bot]