react-native-camera/react-native-camera #2979: iOS 14 video recording wrong aspect ratio/orientation

vrgimael, 2020-09-18:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [X] Did you look for existing matching issues?

**Platforms**
- iOS only

**Versions**

- iOS: 14.0
- react-native-camera: v3.39.1 and master
- react-native: 0.63.2
- react: 16.13.1

**Description/Current Behaviour**

When you record a video, the resulting file's aspect ratio is wrong. It seems like width and height are inverted. Changing camera options like orientation, etc... have no effect.

This only happens on iOS 14.0 - tried with earlier iOS versions and it works fine.

**Expected Behaviour**

Resulting video should match device orientation and viewport.

**Steps to Reproduce**

Set up a new project, record a video file and play it back.

I made a simple camera/playback app here to reproduce:

https://github.com/vrgimael/react-native-camera-test

**Additionals**

This is  what the viewfinder looks like:

[image]

And this is the resulting video. This doesn't happen on versions earlier than iOS 14.

[image]

I'll keep investigating but thought it might be helpful to post it here, anybody has any idea?

Thanks!

vrgimael, 2020-09-18:
Update: same results with expo-camera

ami9000, 2020-09-19:
@vrgimael 
> Update 2: this might be an issue with the video playing lib instead

I think it has to do with the camera library at least. If I inspect a video file recorded under this bug format, the file indeed has the wrong orientation specs - e.g. Recording done in portrait/720p has 1280 width and 720 height instead of the other way around.

vrgimael, 2020-09-19:
So... I have a few more updates here:

First is that if you turn on mirroring, the video strangely behave fine. With this I managed to make a suuuuuper hackish patch in RNCamera.m that "fixes" (or so I thought) the video orientation without actually mirroring.

But then... a few hours later, I realised that somehow even the older videos on my app that were working fine before (are still fine on iOS < 14) are now looking glitched, making me believe the video orientation might somehow be reported wrong.

Weirdly, not all videos are looking distorted. It seems that only the ones uploaded from certain devices are glitched.

When logging the "naturalSize" being reported on my videos, some of them have the wrong width, height and orientation even though the original file is fine on the server... they are not rotated though, only wrong in aspect ratio.

I could "fix" the visual issue by using "stretch" and forcing a certain video size, but since not all of them are reported consistently I cannot predict which ones will be glitched.

Is there any way to check deeper file meta data like width and height? How is this actually reported in the react-native-camera lib?

At this point I'm considering just calculating width and height on the server and giving it to the frontend for layout computation

ami9000, 2020-09-20:
Further analysis: I ran ffprobe on video recorded on two devices, iPhone 7 (iOS 13) and iPhone SE - 2nd gen (iOS 14). Both in portrait 720p. Their metadata has 1280 width and 720 height and *displaymatrix: rotation of -90.00 degrees*. So according to meta data, the two recordings are the same. @vrgimael might be correct that this is a video playing lib issue instead.

jpmazza, 2020-09-21:
This is also happening to me. When I recorded a video from my iPhone 8 with iOS 14 and played the local video that I had just recorded, it looks with this wrong aspect. However, once the video is in the server and I am downloading it to play, it seems with the right layout.

So I'm not sure if this bug comes from this library or react-native-video with local videos. 🤔

vrgimael, 2020-09-21:
So after fighting with this for a while I just ended up cropping the images on the backend, and since my application only uses square images the aspect ratio is irrelevant. So also a note for @jpmazza, my videos are coming from a remote server (not local) and still have the same results.

This doesn't actually solve the problem though. As @ami9000 pointed out, all iOS versions produce the same results on ffprobe indeed with a rotated 90 degrees matrix and inverted width and height.

This probably means 2 things:

1. The react-native-camera seems to be producing metadata incorrectly and it has always been, after all, the videos were taken with the phone upright and when viewed on a computer, it actually looks fine

2. Versions of iOS prior to 14 didn't seem to care if the metadata was wrong and just played the video like desktops do for some reason.

I tried using react-native-video and also react-native-video-player (they might use the same lib in fact), and it wrong on both.

I wonder if this is an issue with the JS library (react-native-video), or maybe change or glitch in Apple's AV api usage that's causing this weird behaviour?

vrgimael, 2020-09-21:
To be honest, there are a few issues related to iOS 14 on [react-native-video](https://github.com/react-native-community/react-native-video/issues?q=is%3Aissue+ios+14+) repo, seems like it broke more than one thing

vrgimael, 2020-09-21:
Update again: it really seems like a video reproduction issue rather than recording.

I just tested iOS 14.2 Beta as pointed out by some users in the react-native-video issues and this ratio problem is gone. Strange, but good I guess

ami9000, 2020-09-23:
There seems to be a potential temporary fix over at react-native-video: https://github.com/react-native-community/react-native-video/issues/2159

I haven't had the chance to test myself yet.

wgy6055, 2020-09-24:
I'm not using react-native-camera, but I'm also confusing with this issue. Is it a bug from Apple?

vrgimael, 2020-09-25:
@wgy6055 can't say for sure it's from Apple but it's fixed on iOS 14.2 Beta as mentioned above, so maybe. Might just be an issue with react-native-video though.

The fix in the issue mentioned by @ami9000 here kinda works but has some weird side effects though as discussed in the thread.

drewandre, 2020-09-25:
Given the amount of video-dimension and other video related issues across similar libraries that have popped up on iOS 14 (and not on android 10/11), this must be an OS level issue right? Here are other issues I gathered:

react-native-camera:
• iOS 14 video recording wrong aspect ratio/orientation
• Recorded Video orientation is buggy
react-native-video:
• Version 4.4.4 IOS14 has the wrong video screen size
• .MOV file portrait video displaying as landscape iOS 14
• Videos not played on the final release of iOS 14
• iOS 14 - Play button disappears resulting in no way to start video playback
expo-video:
• [iOS 14] Video assets does not work
• MOV Video Playback Stretched on iOS
• expo-av Video resizeMode doesn't work on first loop on iOS14

antoarva, 2020-10-21:
Yesterday iOS 14.1 released, the video is fixed now. Just update.

phithu, 2020-10-26:
I also face this issue. My iOS version: 14.1

closed 2021-01-13 by stale[bot]