react-native-camera/react-native-camera #2980: Fatal Exception: java.lang.NullPointerException Attempt to invoke virtual method 'void android.hardware.Camera$Parameters.setRotation(int)' on a null object reference android.view.Orientation

huseyintamer, 2020-09-18:
# Bug Report

**To Do First**

- [no] Did you try latest release?
- [no] Did you try master?
- [no] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: 8,9,10,7
- iOS:
- react-native-camera:3.30.0
- react-native:0.61.5
- react:16.9.0
  

**Description/Current Behaviour**
Many users experience crashes due to this error.

Fatal Exception: java.lang.NullPointerException
Attempt to invoke virtual method 'void android.hardware.Camera$Parameters.setRotation(int)' on a null object reference

**Expected Behaviour**

**Steps to Reproduce**

**Additionals**

[image]

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

vetrivendhan48, 2020-09-23:
Try running 'run-android'

huseyintamer, 2020-09-23:
it happens when I release it.

closed 2020-10-04 by huseyintamer

manohar-octifi, 2020-12-01:
@huseyintamer Did you got any resolution got the same issue on 3.19

```
java.lang.NullPointerException: Attempt to invoke virtual method 'void android.hardware.Camera$Parameters.setRotation(int)' on a null object reference
    at com.google.android.cameraview.Camera1.setDeviceOrientation(Camera1.java:929)
    at com.google.android.cameraview.CameraView$1.onDisplayOrientationChanged(CameraView.java:142)
    at com.google.android.cameraview.DisplayOrientationDetector.dispatchOnDisplayOrientationChanged(DisplayOrientationDetector.java:114)
    at com.google.android.cameraview.DisplayOrientationDetector$1.onOrientationChanged(DisplayOrientationDetector.java:90)
    at android.view.OrientationEventListener$SensorEventListenerImpl.onSensorChanged(OrientationEventListener.java:144)
    at android.hardware.SystemSensorManager$SensorEventQueue.dispatchSensorEvent(SystemSensorManager.java:981)
    at android.os.MessageQueue.nativePollOnce(MessageQueue.java)
    at android.os.MessageQueue.next(MessageQueue.java:326)
    at android.os.Looper.loop(Looper.java:181)
    at android.app.ActivityThread.main(ActivityThread.java:7050)
    at java.lang.reflect.Method.invoke(Method.java)
    at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:494)
    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:965)
```

huseyintamer, 2021-01-18:
@manohar-octifi I update React-Native version to 0.63 and RN-Camera Version to lastest version