react-native-camera/react-native-camera #2983: Autolinking (RN > 0.60) does not work correctly

mrousavy, 2020-09-21:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

iOS

**Versions**

- Android:
- iOS: 14
- react-native-camera: 3.40.0
- react-native: 0.63.2
- react: 16.13.1
  

**Description/Current Behaviour**

When I build my project I get the following warning in my terminal:

```sh
error React Native CLI uses autolinking for native dependencies, but the following modules are linked manually: 
  - react-native-camera (to unlink run: "react-native unlink react-native-camera")
This is likely happening when upgrading React Native from below 0.60 to 0.60 or above. Going forward, you can unlink this dependency via "react-native unlink <dependency>" and it will be included in your app automatically. If a library isn't compatible with autolinking, disregard this message and notify the library maintainers.
Read more about autolinking: https://github.com/react-native-community/cli/blob/master/docs/autolinking.md
```

**Expected Behaviour**

I expect no warning to be shown

**Steps to Reproduce**

1. Install lib into RN > 0.60 project
2. Run

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

mrousavy, 2020-09-21:
There must've been some faulty configuration in the podspec, because this only applies to iOS (I think)

a-eid, 2020-09-22:
I can confirm this happens when adding RNCamera Pods.

closed 2021-01-13 by stale[bot]

anabeatrizzz, 2021-04-22:
I'm having the same issue when `react-native run-ios`.

anabeatrizzz, 2021-04-22:
@MateusAndrade 
`"react-native": "0.64.0"`
`"react-native-camera": "^3.43.0"`

lfrkovic-spaceiq, 2021-07-09:
Same issue:
"react-native": "0.64.1"
"react-native-camera": "^3.43.0"

MateusAndrade, 2021-07-14:
> Same issue:
> "react-native": "0.64.1"
> "react-native-camera": "^3.43.0"

This happens with the `master` version or an previous version?

tanmoythander, 2021-07-27:
```
"react-native": "^0.64.2",
"react-native-camera": "^4.0.0"
```

Still getting this issue while running `react-native run-ios`,
```
error React Native CLI uses autolinking for native dependencies, but the following modules are linked manually:
  - react-native-camera (to unlink run: "react-native unlink react-native-camera")
This is likely happening when upgrading React Native from below 0.60 to 0.60 or above. Going forward, you can unlink this dependency via "react-native unlink <dependency>" and it will be included in your app automatically. If a library isn't compatible with autolinking, disregard this message and notify the library maintainers.
Read more about autolinking: https://github.com/react-native-community/cli/blob/master/docs/autolinking.md
```

tanmoythander, 2021-07-27:
Just found a workaround by disabling auto-linking for `react-native-camera`.
Add the contents below in the file `react-native.config.js` in the root directory
```
module.exports = {
  dependencies: {
    'react-native-camera': {
      platforms: {
        android: null,
        ios: null
      }
    }
  }
};
```
Then run `cd ios && pod install`