react-native-camera/react-native-camera #2985: 'focusDepth' prop has no effect on the Camera component

vetrivendhan48, 2020-09-23:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [ ] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

Tested in Android. No focus length is changed.

**Versions**

- Android: 10
- iOS: -
- react-native-camera: ^3.40.0
- react-native: 0.61.5
- react: 16.9.0

  
    "@react-navigation/bottom-tabs": "^5.5.2",
    "@react-navigation/drawer": "^5.8.2",
    "@react-navigation/native": "^5.7.0",
    "@react-navigation/stack": "^5.9.0",

**Description/Current Behaviour**
Setting the property value 'focusDepth' to 1 will make the camera to focus long located objects.
Setting the property value 'focusDepth' to 0 will make the camera to focus near located objects.

**Expected Behaviour**
Setting the property value 'focusDepth' to 1 still the camera remain in near focus mode.
Setting the property value 'focusDepth' to 0 still the camera remain in near focus mode.

**Steps to Reproduce**

We can reproduce the error with the following snippet.
```
                       <RNCamera
                            focusDepth={0.9}
                            autoFocus={RNCamera.Constants.AutoFocus.off}
                            // autoFocusPointOfInterest={pointOfInterest}
                            ref={cameraRef}
                            style={styles.camera}
                            type={RNCamera.Constants.Type.back}
                            // onGoogleVisionBarcodesDetected={
                            //     disabled ? null : onGoogleVisionBarcodesDetected
                            // }
                            // defaultOnFocusComponent={true}
                        >
                            <BarcodeMask
                                height={80}  width={200}
                                showAnimatedLine={false}
                                backgroundColor={'rgba(0, 0, 0, 0)'}
                            />
                        </RNCamera>

```

Not checked yet

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

vetrivendhan48, 2020-09-24:
On digging the library found the 'focusDepth' is not implemented on the new 'Camera1' Java class. This feature is most important for taking a perfect picture. Need someone to fix this issue.

ferbeab, 2020-09-24:
@vetrivendhan48 I reported this issue 2 months ago and no response yet 😞 . Nice job diving into the code and pinpointing the issue. Let's see if now it can get fixed 💪

vetrivendhan48, 2020-09-30:
Waiting for any response.

vetrivendhan48, 2020-10-09:
Still waiting for a response.

vetrivendhan48, 2021-01-09:
Need confirmation of the issue is fixed from the Maintainers.

closed 2021-03-17 by stale[bot]