react-native-camera/react-native-camera #2986: Consecutive Screens with RNCamera components not working

D4nt4lion, 2020-09-23:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: 29
- react-native-camera: 3.31.1
- react-native: 0.62.1
- react: 16.11.0
- react-navigation/native: 5.1.5
  

**Description/Current Behaviour**

When navigating to a screen with an RNCamera while already having an active RNCamera instance in the parent screen, the Camera on the new Screen stays black and never triggers ready state. For example use Barcode Scanner and then navigate to a Camera Screen for taking a picture.

**Expected Behaviour**

Loading the second camera screen correctly.

**Steps to Reproduce**

Create a Screen with a RNCamera component and then call a new Screen with another RNCamera Component.

**Additionals**

I also tried using useIsFocused hook like suggested here: https://stackoverflow.com/questions/55866696/react-native-camera-works-first-time-then-not-working-black-screen-in-android.
But it did not work.

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

vetrivendhan48, 2020-09-24:
https://github.com/react-native-community/react-native-camera/issues/1671#issuecomment-416872717

D4nt4lion, 2020-09-25:
Yes I am doing that. Still the Camera does not load in my second screen and does not trigger onReady.

SangNguyenAs, 2020-10-02:
Same issue for me. If consecutive screen using RNCamera, then front camera doesn't work (showing black screen) on Samsung Note 8, but still works normally on Xiaomi Redmi Note 8 Pro.

jgenyuk, 2020-10-08:
Same issue for me on Motorola g(7). The log shows that camera is unmounted successfully when passing to the second screen, but it throws an error: "Error mounting camera: Camera view threw an error - component could not be rendered."

Jekkt, 2020-10-21:
I got the same issue.

D4nt4lion, 2020-11-24:
Solution for me was abandoning this dependency and using expo camera instead.
https://docs.expo.io/versions/latest/sdk/camera/

mccraveiro, 2021-01-18:
I'm also having this problem.

@D4nt4lion expo-camera uses an old API and we're having focus issues.

CorentinHeroux, 2021-02-23:
Same issue on OnePlus 5T.

kkx64, 2021-04-28:
Using expo-camera, always getting a black screen on Galaxy XCover 3, works fine on every other phone I tested (Redmi Note 7, Galaxy S8, Huawei P10, Galaxy Note 4). Anyone manage to fix this?

closed 2021-07-04 by stale[bot]

fedeerbes, 2021-09-19:
I'm having the same issue, if it helps to solve this issue, mounting and unmounting RNCamera in the same screen also triggers this issue.

from logcat
```
2021-09-19 15:17:28.372 7594-7594/com.applemoncash D/ViewRootImpl@1bf3928[MainActivity]: ViewPostIme pointer 0
2021-09-19 15:17:28.475 7594-7594/com.applemoncash D/ViewRootImpl@1bf3928[MainActivity]: ViewPostIme pointer 1
2021-09-19 15:17:29.531 7594-7594/com.applemoncash D/ViewRootImpl@1bf3928[MainActivity]: ViewPostIme pointer 0
2021-09-19 15:17:29.595 7594-7594/com.applemoncash D/ViewRootImpl@1bf3928[MainActivity]: ViewPostIme pointer 1
2021-09-19 15:17:30.003 7594-7594/com.applemoncash D/ViewRootImpl@1bf3928[MainActivity]: Relayout returned: old=(0,0,720,1480) new=(0,0,720,1480) req=(720,1480)0 dur=11 res=0x1 s={true 3806078976} ch=false
2021-09-19 15:17:30.792 7594-7594/com.applemoncash D/SensorManager: registerListener :: 0, LIS2DS Accelerometer, 200000, 0,  
2021-09-19 15:17:31.035 7594-7654/com.applemoncash W/Gralloc3: allocator 3.x is not supported
2021-09-19 15:17:31.668 7594-7594/com.applemoncash W/om.applemoncas: Long monitor contention with owner RNCamera-Handler-Thread (7986) at int android.hardware.Camera.native_setup(java.lang.Object, int, int, java.lang.String)(Camera.java:-2) waiters=0 in void com.google.android.cameraview.Camera1.setDisplayOrientation(int) for 850ms
2021-09-19 15:17:31.680 7594-7594/com.applemoncash I/Choreographer: Skipped 51 frames!  The application may be doing too much work on its main thread.
2021-09-19 15:17:33.455 7594-7594/com.applemoncash D/ViewRootImpl@1bf3928[MainActivity]: ViewPostIme pointer 0
2021-09-19 15:17:33.505 7594-7594/com.applemoncash D/ViewRootImpl@1bf3928[MainActivity]: ViewPostIme pointer 1
2021-09-19 15:17:33.833 7594-7594/com.applemoncash D/Camera: app passed NULL surface
2021-09-19 15:17:33.840 7594-7594/com.applemoncash D/SensorManager: unregisterListener ::   
2021-09-19 15:17:33.945 7594-7986/com.applemoncash E/BufferQueueProducer: [SurfaceTexture-0-7594-0] connect: BufferQueue has been abandoned
2021-09-19 15:17:33.945 7594-7986/com.applemoncash E/BufferQueueProducer: [SurfaceTexture-0-7594-0] setMaxDequeuedBufferCount: BufferQueue has been abandoned
2021-09-19 15:17:33.958 7594-7899/com.applemoncash D/SoLoader: About to load: libnative-imagetranscoder.so
2021-09-19 15:17:33.960 7594-7899/com.applemoncash D/SoLoader: libnative-imagetranscoder.so not found on /data/data/com.applemoncash/lib-main
2021-09-19 15:17:33.961 7594-7899/com.applemoncash D/SoLoader: libnative-imagetranscoder.so found on /data/app/com.applemoncash-k-OFnI1R5cp1yHWi226yOA==/lib/arm
2021-09-19 15:17:33.961 7594-7899/com.applemoncash D/SoLoader: Not resolving dependencies for libnative-imagetranscoder.so
2021-09-19 15:17:33.967 7594-7899/com.applemoncash D/SoLoader: Loaded: libnative-imagetranscoder.so
2021-09-19 15:17:34.710 7594-7594/com.applemoncash D/ViewRootImpl@1bf3928[MainActivity]: ViewPostIme pointer 0
2021-09-19 15:17:34.777 7594-7594/com.applemoncash D/ViewRootImpl@1bf3928[MainActivity]: ViewPostIme pointer 1
2021-09-19 15:17:34.787 7594-7621/com.applemoncash W/Camera-JNI: callback on dead camera object
2021-09-19 15:17:35.088 7594-7594/com.applemoncash D/SensorManager: registerListener :: 0, LIS2DS Accelerometer, 200000, 0,  
2021-09-19 15:17:35.160 7594-8104/com.applemoncash E/CAMERA_1::: chooseCamera failed.
    java.lang.RuntimeException: Fail to get camera info
        at android.hardware.Camera._getCameraInfo(Native Method)
        at android.hardware.Camera.getCameraInfo(Camera.java:291)
        at com.google.android.cameraview.Camera1.chooseCamera(Camera1.java:1008)
        at com.google.android.cameraview.Camera1.start(Camera1.java:234)
        at com.google.android.cameraview.CameraView.start(CameraView.java:328)
        at org.reactnative.camera.RNCameraView$4.run(RNCameraView.java:610)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:237)
        at android.os.HandlerThread.run(HandlerThread.java:67)
2021-09-19 15:17:35.194 7594-7697/com.applemoncash I/ReactNativeJS: { message: 'Camera view threw an error - component could not be rendered.' }
2021-09-19 15:17:35.367 7594-7594/com.applemoncash D/SensorManager: unregisterListener ::   
2021-09-19 15:17:35.872 7594-7654/com.applemoncash W/CameraBase: mediaserver's remote binder Camera object died
2021-09-19 15:17:35.872 7594-7654/com.applemoncash W/Camera-JNI: callback on dead camera object
2021-09-19 15:17:35.872 7594-7613/com.applemoncash W/CameraBase: Camera service died!
2021-09-19 15:17:36.356 7594-7594/com.applemoncash D/ViewRootImpl@1bf3928[MainActivity]: ViewPostIme pointer 0
2021-09-19 15:17:36.409 7594-7594/com.applemoncash D/ViewRootImpl@1bf3928[MainActivity]: ViewPostIme pointer 1
2021-09-19 15:17:36.796 7594-7594/com.applemoncash D/SensorManager: registerListener :: 0, LIS2DS Accelerometer, 200000, 0,  
2021-09-19 15:17:36.806 7594-8118/com.applemoncash A/libc: Fatal signal 6 (SIGABRT), code -1 (SI_QUEUE) in tid 8118 (RNCamera-Handle), pid 7594 (om.applemoncash)

```

Platforms
Android

Versions
Android: 29
react-native-camera: 4.2.0
react-native: 0.64.0