react-native-camera/react-native-camera #2987: How to add library for iOS project?

PrayaAmadigaPitasa, 2020-09-26:
# Question

**To Do First**
- [ ] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [ ] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [ ] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**
So, I want to add TensorFlow Lite library for iOS Project, and how to install it? because I can't found the pod file

**Tags**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

rafaelcavalcante, 2020-09-29:
Hi there! I'm using [this library](https://github.com/kaelhem/tflite-react-native) and this fork of [react-native-camera](https://github.com/Boris-c/react-native-camera) to use Tensorflow, but not with much success.

Currently problems:
- As i use TakePictureAsync to get images to use in TensorFlow, the shutter sound keeps playing every N seconds (i'm using an interval)
- I'm trying to figure out how to get a preview image without using TakePictureAsync

Please follow #2984 and #2506

PrayaAmadigaPitasa, 2020-10-01:
> this library

okkay, thank you rafael for the information

closed 2021-01-13 by stale[bot]