react-native-camera/react-native-camera #2989: onFacesDetected breaks app on Android (device-only)

LoverFancy, 2020-09-28:
# Bug Report

**To Do First**

- [yes] Did you try latest release?
- [no] Did you try master?
- [yes] Did you look for existing matching issues?

**Platforms**

android

**Versions**

- Android:10.0
- iOS:
- react-native-camera:3.21.0 and 3.40.0
- react-native:0.61.0
- react:16.9.0

**Description/Current Behaviour**
1. compelete the config according to the document and copy the demo in my app.

//demo code
```
// /* eslint-disable no-console */
import React from 'react';
import { StyleSheet, Text, View, TouchableOpacity, Slider } from 'react-native';
// eslint-disable-next-line import/no-unresolved
import { RNCamera } from 'react-native-camera';

const flashModeOrder = {
  off: 'on',
  on: 'auto',
  auto: 'torch',
  torch: 'off',
};

const wbOrder = {
  auto: 'sunny',
  sunny: 'cloudy',
  cloudy: 'shadow',
  shadow: 'fluorescent',
  fluorescent: 'incandescent',
  incandescent: 'auto',
};

const landmarkSize = 2;

export default class CameraScreen extends React.Component {
  state = {
    flash: 'off',
    zoom: 0,
    autoFocus: 'on',
    depth: 0,
    type: 'front',
    whiteBalance: 'auto',
    ratio: '16:9',
    recordOptions: {
      mute: false,
      maxDuration: 5,
      quality: RNCamera.Constants.VideoQuality['288p'],
    },
    isRecording: false,
    canDetectFaces: false,
    canDetectText: false,
    canDetectBarcode: false,
    faces: [],
    textBlocks: [],
    barcodes: [],
[... 436 lines cut ...]
    borderWidth: 2,
    borderRadius: 2,
    position: 'absolute',
    borderColor: '#F00',
    justifyContent: 'center',
  },
  textBlock: {
    color: '#F00',
    position: 'absolute',
    textAlign: 'center',
    backgroundColor: 'transparent',
  },
});
```
the line in demo
`onFacesDetected={canDetectFaces ? this.facesDetected.bind(this) : null}`
2. after build the app in android devices and open the "onFacesDetected" event, when I open the camera in app and the app crashed.
3. after upgrade the latest react-natice-camera and rebuild it's no work.
4. no any error report when I build it and can success.
5. try to uninstall and install the app no wrok.
6. I comment the line above and the camrea is ok.

**Expected Behaviour**

need to help to fix the bug and keep face detect function ok .

LoverFancy, 2020-10-14:
I found the error info in android studio as below, can help to analysis?

java.lang.RuntimeException: An error occurred while executing doInBackground()
        at android.os.AsyncTask$AsyncFutureTask.done(AsyncTask.java:429)
        at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:383)
        at java.util.concurrent.FutureTask.setException(FutureTask.java:252)
        at java.util.concurrent.FutureTask.run(FutureTask.java:271)
        at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:292)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:929)
     Caused by: java.lang.SecurityException: Failed to find provider com.google.android.gms.phenotype for user 0; expected to find a valid ContentProvider for this authority
        at android.os.Parcel.createException(Parcel.java:2071)
        at android.os.Parcel.readException(Parcel.java:2039)
        at android.os.Parcel.readException(Parcel.java:1987)
        at android.content.IContentService$Stub$Proxy.registerContentObserver(IContentService.java:1234)
        at android.content.ContentResolver.registerContentObserver(ContentResolver.java:2301)
        at android.content.ContentResolver.registerContentObserver(ContentResolver.java:2289)
        at com.google.android.gms.internal.clearcut.zzab.zza(Unknown Source:32)
        at com.google.android.gms.internal.clearcut.zzae.zzl(Unknown Source:34)
        at com.google.android.gms.internal.clearcut.zzae.get(Unknown Source:26)
        at com.google.android.gms.internal.clearcut.zzp.zza(Unknown Source:21)
        at com.google.android.gms.clearcut.ClearcutLogger$LogEventBuilder.log(Unknown Source:74)
        at com.google.android.gms.internal.firebase_ml.zznu.zza(Unknown Source:52)
        at com.google.firebase.ml.vision.face.FirebaseVisionFaceDetector.<init>(Unknown Source:22)
        at com.google.firebase.ml.vision.face.FirebaseVisionFaceDetector.zza(Unknown Source:13)
        at com.google.firebase.ml.vision.FirebaseVision.getVisionFaceDetector(Unknown Source:15)
        at org.reactnative.facedetector.RNFaceDetector.createFaceDetector(RNFaceDetector.java:95)
        at org.reactnative.facedetector.RNFaceDetector.getDetector(RNFaceDetector.java:46)
        at org.reactnative.camera.tasks.FaceDetectorAsyncTask.doInBackground(FaceDetectorAsyncTask.java:76)
        at org.reactnative.camera.tasks.FaceDetectorAsyncTask.doInBackground(FaceDetectorAsyncTask.java:22)
        at android.os.AsyncTask$3.call(AsyncTask.java:389)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:292) 
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167) 
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641) 
        at java.lang.Thread.run(Thread.java:929) 
     Caused by: android.os.RemoteException: Remote stack trace:
        at com.android.server.content.ContentService.registerContentObserver(ContentService.java:351)
        at android.content.IContentService$Stub.onTransact(IContentService.java:482)
        at android.os.Binder.execTransactInternal(Binder.java:1028)
        at android.os.Binder.execTransact(Binder.java:1001)
2020-10-14 17:20:29.927 11258-11258/com.mysoft.yunke.msdjdev E/ResMng NATIVE_MSG_FILTER: endActivityTransaction: margin state not match

LoverFancy, 2020-10-15:
I fix this issue according to the follow idea: https://github.com/onfido/onfido-android-sdk/issues/90

closed 2021-01-13 by stale[bot]

danyalutsevich, 2022-08-18:
have the same problem, any progress?

camboYY, 2022-09-01:
i have the same issue