react-native-camera/react-native-camera #2991: Video is not recording warning

AlkanV, 2020-10-03:
# Bug Report

**To Do First**

- [ Y] Did you try latest release?
- [Y ] Did you try master?
- [ Y] Did you look for existing matching issues?

**Platforms**

Android
iOS 

**Versions**

- Android:
- iOS: 13.5
- react-native-camera: master
- react-native: 0.63.2
- react:16.13.1
- react-navigation:5.7.3

**Description/Current Behaviour**

It is taking pictures as expected but when i implement record video functionalty, it gives vides is not recording yellow warning. i check everywhere but could not find any solution
**Expected Behaviour**
it should record video

**Steps to Reproduce**

**Additionals**

```
import React, { useEffect, useState } from 'react';
import { VideoPropTypes } from '../Navigator/NavigatorTypes';
import { StyleSheet, Text, TouchableOpacity, View, Button } from 'react-native';
import { RecordOptions, RNCamera } from 'react-native-camera';
import { useCamera } from 'react-native-camera-hooks';

const Video: React.FC<VideoPropTypes> = ({ navigation, route }) => {
	const { id, title } = route.params!;
	const [_isRecording, _setIsRecording] = useState(false);
	let camera = React.createRef<RNCamera>();

	const takePicture = async () => {
		if (camera) {
			const options = { quality: 0.5, base64: true };
			const data = await camera.current!.takePictureAsync(options);

			console.log(data.uri);
		}
	};

	const onRecordingStart = () => {
		console.log('RECORDING STARTED');
	  }
	
	  const onRecordingEnd = () => {
		console.log('RECORDING ENDED');
	  }

	const takeVideo = async () => {
		_setIsRecording(true);
		await camera.current!.recordAsync({
			mute: false,
			fps: 60,
		})
			.then((data) => {
				console.log(data);
			})
			.catch((err) => console.log(err))
			.finally(() => _setIsRecording(false));

		// if (camera) {
        //     try {
        //         const options: RecordOptions = {
		// 			quality: "1080p",
		// 			mute: false,
		// 			fps: 60,
		// 			orientation: 'auto'
        //             //videoBitrate: 8000000,
        //             //maxDuration: 30
		// 		};
		// 		console.log('STARTING VIDEO')
        //         const promise = camera.current!.recordAsync(options);
        //         if (promise) {
        //             _setIsRecording(true);
		// 			const data = await promise;
		// 			console.log(data);
		// 			_setIsRecording(false);
		// 			console.log('STOPPING VIDEO')
        //         }
        //     } catch (error) {
        //         console.log(error);
        //     }
		// }
	}

	const stopVideo = async () => {
		await camera.current!.stopRecording();
		_setIsRecording(false);
	}
	return (
		<View style={styles.container}>
			<RNCamera
				ref={camera}
				style={styles.preview}
				type={RNCamera.Constants.Type.back}
				flashMode={RNCamera.Constants.FlashMode.on}
				androidCameraPermissionOptions={{
					title: 'Permission to use camera',
					message: 'We need your permission to use your camera',
					buttonPositive: 'Ok',
					buttonNegative: 'Cancel',
				}}
				androidRecordAudioPermissionOptions={{
					title: 'Permission to use audio recording',
					message: 'We need your permission to use your audio',
					buttonPositive: 'Ok',
					buttonNegative: 'Cancel',
				}}
				onGoogleVisionBarcodesDetected={({ barcodes }) => {
					console.log(barcodes);
				}}
				onRecordingStart={onRecordingStart}
				onRecordingEnd={onRecordingEnd}
			/>
			<View style={{ flex: 0, flexDirection: 'row', justifyContent: 'center' }}>
				<TouchableOpacity onPress={() => takePicture()} style={styles.capture}>
					<Text style={{ fontSize: 14 }}> SNAP </Text>
				</TouchableOpacity>
				<TouchableOpacity onPress={() => _isRecording ? takeVideo() : stopVideo()} style={styles.capture}>
					<Text style={{ fontSize: 14 }}> {_isRecording ? 'stop' : 'start'} </Text>
				</TouchableOpacity>
			</View>
		</View>
	);
};

const styles = StyleSheet.create({
	container: {
		flex: 1,
		flexDirection: 'column',
		backgroundColor: 'black',
	},
	preview: {
		flex: 1,
		justifyContent: 'flex-end',
		alignItems: 'center',
	},
	capture: {
		flex: 0,
		backgroundColor: '#fff',
		borderRadius: 5,
		padding: 15,
		paddingHorizontal: 20,
		alignSelf: 'center',
		margin: 20,
	},
});

export default Video;

```

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

closed 2020-10-28 by AlkanV

gsavvid, 2020-11-03:
Why did you close this @AlkanV ? Can you post the solution if you've found one? I'm having the same issue.

zendevil, 2021-01-19:
I have the same issue.

elleryfamilia, 2021-07-16:
Running into this same issue. Anyone figure it out?

nirmalrathore, 2021-12-24:
I have same issue

nirmalrathore, 2021-12-24:
Please explain me if any one find any solution