react-native-camera/react-native-camera #2993: mlkit Example not working on iOS

heraldovalenti, 2020-10-06:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

iOS

**Versions**

- iOS: 14.0.0
- react-native-camera: 3.40.0 / master
- react-native: 0.63.2
- react: 16.13.1

**Description/Current Behaviour**

MLKit example for iOS is broken - it doesn't run.
TL;DR: 

**Expected Behaviour**

MLKit example to run out of the box following `examples/mlkit/README.md` instructions, which are:
1. Run `yarn install`.
2. Create Firebase project, generate `google-services.json` and place it into `./android/app` folder, generate `GoogleService-Info.plist` and place it into `./ios/mlkit` folder.
3. Build project (you will likely need to manage signing if you are building for ios device)

**Steps to Reproduce**
1. Clone this repository and checkout `master` branch.
2. Move to the `examples/mlkit` folder: `cd examples/mlkit`.
3. Install NPM dependencies: `yarn install`.
4. [Fails] Install Pod dependencies: `cd ios && pod install && cd ..`.
It complains about wrong pod configs:
```
Analyzing dependencies
Fetching podspec for `DoubleConversion` from `../node_modules/react-native/third-party-podspecs/DoubleConversion.podspec`
Fetching podspec for `Folly` from `../node_modules/react-native/third-party-podspecs/Folly.podspec`
Fetching podspec for `glog` from `../node_modules/react-native/third-party-podspecs/glog.podspec`
[!] The name of the given podspec `Yoga` doesn't match the expected one `yoga`

[!] Automatically assigning platform `iOS` with version `9.0` on target `mlkit` because no platform was specified. Please specify a platform for this target in your Podfile. See `https://guides.cocoapods.org/syntax/podfile.html#platform`.
```
These can be fixed by updating the Podfile with latest react-native configurations, something like the following:
```
require_relative '../node_modules/react-native/scripts/react_native_pods'
require_relative '../node_modules/@react-native-community/cli-platform-ios/native_modules'

platform :ios, '10.0'

target 'mlkit' do
  config = use_native_modules!

  use_react_native!(:path => config["reactNativePath"])

  target 'mlkitTests' do
    inherit! :complete
  end

  use_flipper!
  post_install do |installer|
    flipper_post_install(installer)
  end

  pod 'react-native-camera', path: '../../../', subspecs: [
    'TextDetector',
    'FaceDetectorMLKit',
    'BarcodeDetectorMLKit'
  ]

  pod 'Firebase/Core'
end

target 'mlkit-tvOS' do

  target 'mlkit-tvOSTests' do
    inherit! :search_paths
  end
end

```
Then removing the `Podfile.lock` and retrying `pod install` should succeed. And then, next steps would be:
  5. Open the `mlkit.xcworkspace` with XCode: `open mlkit.xcworkspace`.
  6. Fix the signing issues that might come up.
  7. Add a `GoogleService-Info.plist` file under `ios/mlkit` folder.
  8. [Fail] Build the project. Fails becuase of dependencies, links and build configurations. Couldn't make it work.

I understand the root cause of this issue is because of obsolete iOS project configuration for newer version of react-native. So, I went ahead and:
1. Created a new react-native project with latest 0.63.3
2. Merged new configurations with the example

At this point the XCode project is able to run on my device but the bundler fails trying to solve some modules (first babel, then the react-native-camera itself). So:

  3. Added the latest react-native-camera dependency directly on the mlkit example: `yarn add react-native-camera@3.40.0`
  4. Updated the metro configuration file `metro.config.js` to remove the dependency resolution from the root folder.

Finally with this, got the example working :)
I can submit a PR with the changes to fix it, and also try it on Android because I'm not sure if the same issue is present, but before I would like to ask:
Has anyone faced this issue recently? Are the mlkit/basic examples still maintained with latest changes? Does it work for you without adding the react-native-camera and pulling it from the root repository?
I see some recent changes on the examples but I would like to know if they are working for the rest of the users, in which case: how do I run the project out-of-the box? Do you have updated instructions?
Thanks a lot!

zn7726, 2020-11-13:
I came across the same issues as you've got!

I firstly added RNCamra to my existing project and can build in XCode and my app could run. I then cloned the repo to see if the examples are good to run on devices. However, the (same) nightmare began. 

I'll do what you eventually did (creating a new project and add things in). Thanks for sharing your experience.

closed 2021-01-19 by stale[bot]

seancheung, 2021-07-09:
the example itself has been broken for a long time