react-native-camera/react-native-camera #2994: Stretched video when recording in portrait mode (ios)

yesterday24, 2020-10-07:
I have a problem on IOS 14.0, RN version 0.63.2.

When using RNCamera and recording a video in **portrait mode**, the resulting video is stretched. When recording in **landscape mode**, everything works fine.

I've included some screenshots from my device.

[image]

Screenshot 1: video recorded in portrait mode. The video (as seen in the screenshot above) is stretched horizontally, while it should be more of a rectangle (height > width), so stretched vertically, not horizontally.

[image]

Screenshot 2: this video is recorded in landscape mode. The video is not stetched but is rotated 90°.

How can I fix these issues, or is this a bug? It's weird, because it worked before.

Cristi6, 2020-10-28:
Having the same issue, looking forward for a fix or at least a quick solution

wxsms, 2020-12-02:
I have meet the same issue, but only with some ios devices, not all of them.

wxsms, 2020-12-02:
maybe duplicated with #2979

closed 2021-02-07 by stale[bot]