react-native-camera/react-native-camera #2995: Video thumbnail not loading using <Image>

gsavvid, 2020-10-08:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [ ] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

iOS

**Versions**

- iOS: 13.7
- react-native-camera: 3.38.0
- react-native: 0.63.2
- react: 16.13.1
  

**Description/Current Behaviour**

After recording a video I get a uri that looks like this: 
```
file:///var/mobile/Containers/Data/Application/0AB78585-1852-4546-99E6-E3EB68F04799/Library/Caches/Camera/63253D57-7C10-4C17-B312-A2C9D3A9AFE1.mov
``` 
If I set this uri as the `<Image>` source, I get the following error: 

```
Error decoding image data <NSData 0x28058b8a0; 380900 bytes>
```
**Expected Behaviour**

We should be able to set this uri as the source of an `<Image>` component and display a thumbnail preview.

**Steps to Reproduce**

Record a video and then set the produced uri to an `<Image>`:

```
<Image 
   style={styles.image} 
   source={{uri: uri}} 
   onError={(error) => console.log('Error: ' + JSON.stringify(error.nativeEvent.error))} />
```

rafaelcavalcante, 2020-10-08:
Hi @gsavvid!

Please check the React Native documentation for the [Image component](https://reactnative.dev/docs/image.html#source) support file formats. MOV is not supported.

closed 2020-10-08 by rafaelcavalcante

gsavvid, 2020-10-08:
Hi @rafaelcavalcante, thanks for the quick response. 

That doesn't seem right, though. The following `.mov` file loads just fine:

```
assets-library://asset/asset.mov?id=2301F19D-C3DC-4BB4-9E2C-DEDCA52EDBC8&ext=mov
```

Other non-image formats that are not listed in that link work, fine, too, (eg. `.m3u8` files). 

Perhaps the issue is not with react-native-camera but with `<Image>`. However, I was wondering since there's in fact a way to load `.mov` files, is it possible to get the file/path in a way that would be supported by `<Image>`? Something in the format of `assets-library://` for instance?

rafaelcavalcante, 2020-10-08:
Hi @gsavvid. Im my projects i use [react-native-video](https://github.com/react-native-video/react-native-video), but i can't tell you if it supports .mov format, because i always use .mp4. I believe the path provided in your first comment (file://...) will work fine.

And you're right, this is not a problem with react-native-camera.