react-native-camera/react-native-camera #2996: Support IOS multiple cameras

mk-nickyang, 2020-10-09:
# Feature-Request

**Describe the Feature**
Ability to select which back camera (e.g. wide angle lenses) to use for IOS devices (iOS 10+)

**Possible Implementations**
This is implemented for Android in https://github.com/react-native-community/react-native-camera/pull/2492. Similar approach can apply to IOS
- Users get all available cameras using [`getCameraIdsAsync`](https://react-native-community.github.io/react-native-camera/docs/rncamera#getcameraidsasync-promise)
- Find out the ID of the camera they want
- Pass it to [`cameraId`](https://react-native-community.github.io/react-native-camera/docs/rncamera#android-cameraid) (Android Only currently)

**Related Issues**
https://github.com/react-native-community/react-native-camera/issues/2379
Android Implementation PR: https://github.com/react-native-community/react-native-camera/pull/2492

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this feature to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

StephaneJuban, 2020-11-02:
Isn't it already release since v3.7.0 ?
https://github.com/react-native-camera/react-native-camera/pull/2510

For me on v3.40.0, on iOS I can get all camera IDs available using `getCameraIdsAsync` and choose the one that I want using the `cameraId` property

ziiw, 2020-11-20:
> Isn't it already release since v3.7.0 ?
> #2510
> 
> For me on v3.40.0, on iOS I can get all camera IDs available using `getCameraIdsAsync` and choose the one that I want using the `cameraId` property

It's not exactly what the feature is describing : "Ability to select which back camera (e.g. wide angle lenses)".

The #2510 is only returning a bunch of IDs but you have no details about the type of camera available. So you can't let the user choose a type of camera, you can just roll through them.

closed 2021-01-27 by stale[bot]