react-native-camera/react-native-camera #2997: Problem saving photo, with a filter on the face

fera765, 2020-10-11:
I mounting a lib to detect a face and add filters, but when saving the image the filter is disappearing ....

As an alternative I tried to use the react-native-view-shot but I got the same problem, I realized that when I used the camera it excels over the other components

doranteseduardo, 2020-10-16:
Which library are you using?
If the filters are an overlap then they're never going to be part of the captured picture.

fera765, 2020-10-16:
> Which library are you using?
> If the filters are an overlap then they're never going to be part of the captured picture.

I'm using react-native-view-shot, to try to capture the photo ...

However, in all the methods available for this lib I was not successful...
I have the same problem in this link below where a developer faced the same problems

https://github.com/react-native-camera/react-native-camera/issues/2035

doranteseduardo, 2020-10-16:
I meant the filters library.
I've checked the view-shot library, it has an interoperability table on its README, it seems to work well with react-native-camera on Android but not on iOS.

fera765, 2020-10-16:
> I meant the filters library.
> I've checked the view-shot library, it has an interoperability table on its README, it seems to work well with react-native-camera on Android but not on iOS.

I'm assembling the effects by hand, without using third libraries ....

But even when I try to capture the photo with a simple <Text>, it doesn't work either.
These two photos also show some Views, which does not appear when capturing the photo ...
Try using position absolute, and relative, zindex and nothing caught,
I even looked for some other lib of camera with facial recognition, but I didn't find anything nice ...

[image]
[image]

doranteseduardo, 2020-10-16:
Well, that's unfortunate, but in terms of react-native-camera, it won't get your filters rendered on the saved picture because they are not part of the picture.

Using react-native-view-shot seems to be the most feasible option for now, but as it doesn't work well capturing the camera view on iOS, you could try the following approach:

- Take your photo using react-native-camera.
- After your photo is captured, show the picture in the fullscreen.
- Put your filters above the picture.
- Use react-native-view-shot to capture that view.

Tried it on Android 10 and worked well.

fera765, 2020-10-16:
> Well, that's unfortunate, but in terms of react-native-camera, it won't get your filters rendered on the saved picture because they are not part of the picture.
>
> Using react-native-view-shot seems [...]

I also thought about this possibility to capture the image, and then use a new library to identify the user's face and then apply the filter ...... But I thought it would be a little strange for my purpose 🤪

I had found react native opencv 3, however it only works in landscaoe mode and I didn't like it very much

fera765, 2020-10-16:
react-native-opencv3 is a nice lib, but it has some problems and is no longer maintained by the project owner

closed 2021-01-13 by stale[bot]

vannguyendev96, 2021-06-07:
did you fix this error? @fera765

wiredmatrix, 2021-06-17:
Any solution for this issue?
My filters are not saving after recording a video with RNCamera.