react-native-camera/react-native-camera #2998: When I open the face detection function app crash in android

LoverFancy, 2020-10-14:
2020-10-14 17:20:21.578 11258-11258/? E/t.yunke.msdjde: Unknown bits set in runtime_flags: 0x8000
2020-10-14 17:20:21.828 11258-15988/com.mysoft.yunke.msdjdev E/ActivityThread: Failed to find provider info for com.google.android.gms.chimera
2020-10-14 17:20:21.842 11258-15994/com.mysoft.yunke.msdjdev E/Parcel: Reading a NULL string not supported here.
2020-10-14 17:20:29.921 11258-16018/com.mysoft.yunke.msdjdev E/AndroidRuntime: FATAL EXCEPTION: AsyncTask #1
    Process: com.mysoft.yunke.msdjdev, PID: 11258
    java.lang.RuntimeException: An error occurred while executing doInBackground()
        at android.os.AsyncTask$AsyncFutureTask.done(AsyncTask.java:429)
        at java.util.concurrent.FutureTask.finishCompletion(FutureTask.java:383)
        at java.util.concurrent.FutureTask.setException(FutureTask.java:252)
        at java.util.concurrent.FutureTask.run(FutureTask.java:271)
        at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:292)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:929)
     Caused by: java.lang.SecurityException: Failed to find provider com.google.android.gms.phenotype for user 0; expected to find a valid ContentProvider for this authority
        at android.os.Parcel.createException(Parcel.java:2071)
        at android.os.Parcel.readException(Parcel.java:2039)
        at android.os.Parcel.readException(Parcel.java:1987)
        at android.content.IContentService$Stub$Proxy.registerContentObserver(IContentService.java:1234)
        at android.content.ContentResolver.registerContentObserver(ContentResolver.java:2301)
        at android.content.ContentResolver.registerContentObserver(ContentResolver.java:2289)
        at com.google.android.gms.internal.clearcut.zzab.zza(Unknown Source:32)
        at com.google.android.gms.internal.clearcut.zzae.zzl(Unknown Source:34)
        at com.google.android.gms.internal.clearcut.zzae.get(Unknown Source:26)
        at com.google.android.gms.internal.clearcut.zzp.zza(Unknown Source:21)
        at com.google.android.gms.clearcut.ClearcutLogger$LogEventBuilder.log(Unknown Source:74)
        at com.google.android.gms.internal.firebase_ml.zznu.zza(Unknown Source:52)
        at com.google.firebase.ml.vision.face.FirebaseVisionFaceDetector.<init>(Unknown Source:22)
        at com.google.firebase.ml.vision.face.FirebaseVisionFaceDetector.zza(Unknown Source:13)
        at com.google.firebase.ml.vision.FirebaseVision.getVisionFaceDetector(Unknown Source:15)
        at org.reactnative.facedetector.RNFaceDetector.createFaceDetector(RNFaceDetector.java:95)
        at org.reactnative.facedetector.RNFaceDetector.getDetector(RNFaceDetector.java:46)
        at org.reactnative.camera.tasks.FaceDetectorAsyncTask.doInBackground(FaceDetectorAsyncTask.java:76)
        at org.reactnative.camera.tasks.FaceDetectorAsyncTask.doInBackground(FaceDetectorAsyncTask.java:22)
        at android.os.AsyncTask$3.call(AsyncTask.java:389)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at android.os.AsyncTask$SerialExecutor$1.run(AsyncTask.java:292) 
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167) 
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641) 
        at java.lang.Thread.run(Thread.java:929) 
     Caused by: android.os.RemoteException: Remote stack trace:
        at com.android.server.content.ContentService.registerContentObserver(ContentService.java:351)
        at android.content.IContentService$Stub.onTransact(IContentService.java:482)
        at android.os.Binder.execTransactInternal(Binder.java:1028)
        at android.os.Binder.execTransact(Binder.java:1001)
2020-10-14 17:20:29.927 11258-11258/com.mysoft.yunke.msdjdev E/ResMng NATIVE_MSG_FILTER: endActivityTransaction: margin state not match

closed 2020-10-15 by LoverFancy

fhqfhq, 2021-06-19:
你好，遇到同样的问题，请问你是如何解决的 @LoverFancy

LoverFancy, 2021-07-19:
@fhqfhq 
有个hack方法,临时解决：https://github.com/react-native-camera/react-native-camera/issues/2989