react-native-camera/react-native-camera #3000: Looking for new maintainers

sibelius, 2020-10-14:
I'm stepping down from maintaining react-native-camera

Please comment if you wanna maintain this package

Another direction would be deprecated this package (https://github.com/react-native-community/react-native-camera/issues/2942) and focus on expo-camera instead

fabriziobertoglio1987, 2020-10-14:
This weekend I will give a look at the issue. Thanks

doranteseduardo, 2020-10-14:
Considering that a lot of people and projects use this library, I'm sure this project must be kept alive.
I'll talk with my team, I'm sure we can maintain it.

MateusAndrade, 2020-10-15:
I can help with that, but it would be amazing having someone focused on this project.

mateosilguero, 2020-10-16:
Could i help ?

msriaz, 2020-10-18:
I would love to help in maintaining this package.

diegotsi, 2020-10-23:
Hi @sibelius! I can help!

np6176a, 2020-10-29:
Me too!! @sibelius I would love to help maintain this project.

sibelius, 2020-10-29:
to start review and answer or close issues and pull requests

merge pull requests that make sense after testing in android and ios device

Yandamuri, 2020-11-10:
We are starting new project and thinking to use this package. Can we go ahead with this package or is it going to be stopped in the future?

renatobenks, 2020-11-19:
@Yandamuri you can go ahead! it will just be stopped for while until be completely re-designed, in order to keep it supportable

Yandamuri, 2020-11-20:
@renatobenks that's great to hear. Thanks  a lot.

mattonit, 2020-12-05:
I hope it's not too late :) If you're still looking for maintainers, I'd be happy to help with keeping this repo alive

russelRajitha, 2020-12-18:
I am using this app for lot of my projects. i'am really happy to be a contributor with this module and i see some issues like crashing android app are still open.

Walesson, 2020-12-30:
I hope it's not too late, i wish i could help.

mattonit, 2020-12-30:
@sibelius you've added a lot of new maintainers - can we close this issue or you are going to extend this group?

tewarig, 2021-01-02:
well, I  started building apps with react-native in the lockdown of 2020, this package helped me a lot. I would like to give my free time to this.

MarcoScabbiolo, 2021-01-11:
Hey @sibelius i'd be glad to help!

tewarig, 2021-01-12:
hi, @sibelius I would also love to maintain this project.

bkvirendra, 2021-01-15:
@sibelius, we have been using this package for a while now and would be more than happy to assist with further development.

MarcoScabbiolo, 2021-02-08:
Thanks to everyone :heart: 
All hands on deck for the next version!!! #3100

geonsangyoo, 2021-02-09:
@MarcoScabbiolo 
I always appreciate your team for making us to be able to use react-native-camera library. 😄 

Hopefully it is not too late, Is there a seat available as a maintainer?
I would like to onboard on your project and contribute to it more as much as I can, which would be my honor! ^^

Thank you 👍

theasc, 2021-02-11:
I would also like to help :)

ZacharyLangley, 2021-03-11:
The company I work for uses this package and I would like to be added as a maintainer.

MarcoScabbiolo, 2021-03-11:
To anyone looking to contribute, please do so, permissions will be granted as they are needed, nothin prevents you from answering issues, and creating PRs.

fabriziobertoglio1987, 2021-03-31:
can we have stalebot removed?

mayank-budhiraja, 2021-04-14:
I am happy to contribute to the project.

ShakeelAhmad-dev, 2021-08-03:
it's really good package ...
I want to contribute ...

pke, 2021-09-27:
I would be able to help maintain this repo. At least merge fully tested PRs once in a while.
Also would bring in support for volume buttons as per #764 using non-obstrusive feature detection code (when the react native button bundle is available)

pke, 2021-10-04:
> To anyone looking to contribute, please do so, permissions will be granted as they are needed, nothin prevents you from answering issues, and creating PRs.

@MarcoScabbiolo well PRs are piling up already. What prevents them from being merged? Without a clear direction it does not seem to make much sense to work on PRs if they get stale here.

For instance the lib still uses outdated ref handling. This could probably patched easily. I am working on it to patch it locally, since the current implementation does not allow the RNCamer to be mocked properly. But why maintain a PR alongside my branch when its never merged into main here?

MarcoScabbiolo, 2021-10-04:
@pke Sorry if any other issues or discussions are misleading, I've already expressed my mind about this repo and stopped working on the v4 version. My recommendation for you and anyone else is to migrate to [react-native-vision-camera](https://github.com/mrousavy/react-native-vision-camera). There is also [a discussion](https://github.com/mrousavy/react-native-vision-camera) about making it the official successor of `react-native-camera`. If it was only up to me, I would officialy deprecate `react-native-camera` and encourage users to move to `react-native-vision-camera` but I'm not the only one calling the shots here and honestly I've been out of RN in general for while now.

pke, 2021-10-04:
@MarcoScabbiolo oh very cool! I was not aware of the vision camera. And it works in Android too it seems. I like the frame snippets to integrate with frame analysing tooling.
I'll give it a try, thanks!

closed 2021-12-10 by stale[bot]

kidculli, 2022-01-07:
I have tried using react-native-vision-camera however it is definitely not fit for production use. I encounter bugs that would cause the my application to crash. react-native-camera is much more stable. We should at the very least revert the readme back instead of pushing people to use  react-native-vision-camera.

ahron1, 2022-01-09:
It seems there are many volunteers to maintain this. So why is it deprecated? A lot of people are already using this in prod. Why make them migrate if there's no shortage of maintainers?

MarcoScabbiolo, 2022-01-11:
There is shortage of maintainability, not maintainers. It is deprecated because for several reasons the base code is unsustainable, specially in Android. The end-game of restructuring, refactoring and taking the project to a sustainable status is the same end-game react-native-vision-camera is aiming for, and is well on track to reach and way ahead of react-native-camera-v4 partly because it doesn't have to deal with the multiple backwards compatibility issues and braking changes that arise when attempting to refactor from the react-native-camera code, and mainly because it is under very active development.

This is why this has been deprecated and why react-native-vision-camera has been pointed out as the most viable successor. It doesn't mean it's the only option, there are others but their foundations are different. React-native-camera-kit is targeted to a specific use case (Tesla's) and expo's is well... Expo, not RN. What react-native-camera was to me and probably many of you too, is the general-purpose RN library to integrate with the native camera/s. And as I've said many times before, Expo native modules should expand RN ones adding their own Expo stuff, because Expo is built on top of RN, so Expo modules should be built on top of RN ones. It is a waste of resources to have a dedicated from-scratch camera module only for expo, when all of those resources could be dedicated to create a RN module compatible with Expo.

The deprecation notice really means: Keep using react-native-camera if you want, but all the people that have been working on it have agreed that the project is unsustainable and should be replaced. And we luckily have a successor to put our efforts into, to make it the next react-native-camera we can all use for many purposes and collaborate on.

And don't forget this is a community driven project, if you wan't to keep using react-native-camera and contribute on it you can resume the react-native-camera-v4 project or you can fork react-native-camera and create another potential module to use.

Mattimus333, 2022-02-22:
> I have tried using react-native-vision-camera however it is definitely not fit for production use. I encountered bugs that would cause my application to crash. react-native-camera is much more stable. We should at the very least revert the readme back instead of pushing people to use react-native-vision-camera.
> 
> Also we should advertise additional options
> 
> * https://github.com/teslamotors/react-native-camera-kit
> * https://docs.expo.dev/versions/latest/sdk/camera/

Thankyou for this.  React-native-vision-camera doesn't even npm install properly.  The "example" is straight up unusable it's so buggy.  Can't believe that's what this repo is suggesting people switch over to.  react-native-camera-kit looks pretty good.

nes123, 2022-08-20:
> Thank you for this. React-native-vision-camera doesn't even npm install properly. The "example" is straight up unusable it's so buggy. Can't believe that's what this repo is suggesting people switch [...]

React-native-camera-kit seems weak for the android use case. Any thoughts on that?

mat2718, 2022-09-26:
@sibelius  i know that this library was discontinued. is there anyway i could become a maintainer for this library?

ahron1, 2022-09-28:
> @sibelius i know that this library was discontinued. is there anyway i could become a maintainer for this library?

https://github.com/react-native-camera/react-native-camera/issues/3000#issuecomment-1010283718