react-native-camera/react-native-camera #3008: Use of undeclared identifier 'FIRApp' in iOS when using TextRecognition

tmoorthymca, 2020-10-24:
I have tried to using below react-native-camera for **TextRecognition** in iOS.

***react-native-camera-3.40.0***

Its working when  creating dev build. but its not working for Archive(Release Build). In Android its working fine. In iOS I got below error message.

**"Use of undeclared identifier 'FIRApp'"**  in **AppDelegation.m** @[FIRApp configure];

I updated Cocoa-pods and **Firebase(6.34.0)** lates version but still the issue is present. Please help us resolve the issue.

sahinbezeng, 2020-11-13:
In AppDelegate.m file, you should import firebase like `#import <Firebase.h>` before `#ifdef FB_SONARKIT_ENABLED` line.

As I understand, You probably get success on dev build because FlipperKit libraries are used for debugging and this `#ifdef FB_SONARKIT_ENABLED` if condition is satisfied. When you try to archive it will not be imported and the variables will be undeclared because `#import <Firebase.h>` remains in that if condition.

I got the same problem and it was solved in this way.

closed 2021-01-12 by MarcoScabbiolo

jyu520, 2021-04-15:
> In AppDelegate.m file, you should import firebase like `#import <Firebase.h>` before `#ifdef FB_SONARKIT_ENABLED` line.
>
> As I understand, You probably get success on dev build because FlipperKit [...]

Someone needs to add this to the docs!

chrismazzeo, 2021-05-11:
totally agree, happen the same to me, thanks for the answer @jyu520

sakthiinfotec, 2021-06-22:
> Someone needs to add this to the docs!

Thanks @jyu520! The solution worked for me

aymkin, 2021-09-08:
Also, helpful use condition, ad mentioned [here](https://rnfirebase.io/#configure-firebase-with-ios-credentials)

```swift
if ([FIRApp defaultApp] == nil) {
      [FIRApp configure];
    }
```

aliraza96, 2021-12-27:
> What work for me,
> Just add
> `#import <Firebase.h>` to the AppDelegate.h as well

Perfect Solution. Worked for me. Thanks

surajprasad13, 2022-01-06:
> What work for me, Just add `#import <Firebase.h>` to the AppDelegate.h as well

Thanks for saving my time

sarimahmad, 2022-03-20:
> 

You are my Loveee ❤️❤️❤️❤️❤️

sarimahmad, 2022-03-20:
> What work for me,
> Just add
> `#import <Firebase.h>` to the AppDelegate.h as well

Loveee you bro Saved my time ❤️

criticalmiind, 2022-03-24:
ISSUE: undeclared identifier 'FIRApp' in iOS 

Solution:
AppDelegate.m

`
#import "AppDelegate.h"

#import <React/RCTBridge.h>
#import <React/RCTBundleURLProvider.h>
#import <React/RCTRootView.h>

// Add Line Must Before #ifdef FB_SONARKIT_ENABLED
#import <Firebase.h>

#ifdef FB_SONARKIT_ENABLED
#import <FlipperKit/FlipperClient.h>
#import <FlipperKitLayoutPlugin/FlipperKitLayoutPlugin.h>
#import <FlipperKitUserDefaultsPlugin/FKUserDefaultsPlugin.h>
#import <FlipperKitNetworkPlugin/FlipperKitNetworkPlugin.h>
#import <SKIOSNetworkPlugin/SKIOSNetworkAdapter.h>
#import <FlipperKitReactPlugin/FlipperKitReactPlugin.h>

static void InitializeFlipper(UIApplication *application) {
  FlipperClient *client = [FlipperClient sharedClient];
  SKDescriptorMapper *layoutDescriptorMapper = [[SKDescriptorMapper alloc] initWithDefaults];
  [client addPlugin:[[FlipperKitLayoutPlugin alloc] initWithRootNode:application withDescriptorMapper:layoutDescriptorMapper]];
  [client addPlugin:[[FKUserDefaultsPlugin alloc] initWithSuiteName:nil]];
  [client addPlugin:[FlipperKitReactPlugin new]];
  [client addPlugin:[[FlipperKitNetworkPlugin alloc] initWithNetworkAdapter:[SKIOSNetworkAdapter new]]];
  [client start];
}
#endif

@implementation AppDelegate

- (BOOL)application:(UIApplication *)application didFinishLaunchingWithOptions:(NSDictionary *)launchOptions
{

// Add This Condition
if ([FIRApp defaultApp] == nil) {
  [FIRApp configure];
}
// End

#ifdef FB_SONARKIT_ENABLED
  InitializeFlipper(application);
#endif

  RCTBridge *bridge = [[RCTBridge alloc] initWithDelegate:self launchOptions:launchOptions];
  RCTRootView *rootView = [[RCTRootView alloc] initWithBridge:bridge
                                                   moduleName:@"Rider"
                                            initialProperties:nil];

  if (@available(iOS 13.0, *)) {
      rootView.backgroundColor = [UIColor systemBackgroundColor];
  } else {
      rootView.backgroundColor = [UIColor whiteColor];
  }

  self.window = [[UIWindow alloc] initWithFrame:[UIScreen mainScreen].bounds];
  UIViewController *rootViewController = [UIViewController new];
  rootViewController.view = rootView;
  self.window.rootViewController = rootViewController;
  [self.window makeKeyAndVisible];
  return YES;
}

- (NSURL *)sourceURLForBridge:(RCTBridge *)bridge
{
#if DEBUG
  return [[RCTBundleURLProvider sharedSettings] jsBundleURLForBundleRoot:@"index" fallbackResource:nil];
#else
  return [[NSBundle mainBundle] URLForResource:@"main" withExtension:@"jsbundle"];
#endif
}

@end

`

suhaspathak, 2022-12-13:
> What work for me,
> Just add
> `#import <Firebase.h>` to the AppDelegate.h as well

It is work for me. save lot of time. thanks

evangrim, 2023-02-03:
Make sure your `#import <Firebase.h>` is not inappropriately placed in an if block . :)

I accidentally put `#import <Firebase.h>` within `#if RCT_NEW_ARCH_ENABLED` block while upgrading;  I haven't enabled the new arch, so Firebase didn't load.