react-native-camera/react-native-camera #3010: Capture Photo is too slow in Android when fixOrientation: true

retyui, 2020-10-27:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

Android only

**Versions**

- Android: 29
- iOS: ...
- react-native-camera: 3.40.0
- react-native: 0.60.4
- react: 16.9.0

**Description/Current Behaviour**

|  fixOrientation: true | fixOrientation: false |
|---|---|
| [image] | [image] |

**Expected Behaviour**

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

cristianoccazinsp, 2020-11-12:
`fixOrientation` adds a lot of processing. Any setting that alters the original image's data, will require the bitmap to be loaded into memory, be processed, and then saved again. Blame Android for being so slow at doing this.

I suggest you don't do any processing through the camera library, but rather use a resize library (e.g., react-native-image-resizer) afterwards and do the post-processing asynchronously.

retyui, 2020-11-13:
I found a workaround to fix [`EXIF Orientation`](https://www.impulseadventure.com/photo/exif-orientation.html)

[image]

---

First of all disable `fixOrientation: false` than before send\save image to server

using a  `react-native-image-resizer` module resize an image  with `rotation = 0` 

That option on Android apply additional orientation: 

[ImageResizer.java#L516-L521](https://github.com/bamlab/react-native-image-resizer/blob/64ebc6a2fd6aaf2c1ccf9d4d25f056c4268d7ae1/android/src/main/java/fr/bamlab/rnimageresizer/ImageResizer.java#L516-L521)

```java
        // Rotate if necessary. Rotate first because we will otherwise
        // get wrong dimensions if we want the new dimensions to be after rotation.
        // NOTE: This will "fix" the image using it's exif info if it is rotated as well.
        Bitmap rotatedImage = sourceImage;
        int orientation = getOrientation(context, imageUri);
        rotation = orientation + rotation;
```

closed 2020-11-13 by retyui

cristianoccazinsp, 2020-11-13:
Just curious, using capture + react-native-image-resizer isn't as slow as just using capture with `fixOrientation: true`? 

I know there's a huge advantage when using react-native-image-resizer because it can be done asynchronously after a capture (and it doesn't block the UI thread) so there's the illusion of photos getting captured faster, but overall the result should be the same.

retyui, 2020-11-13:
For our case, I had to fix the Exif orientation before sending it to the server. 

So users taking high-quality photos and it works quickly. 

and only when he press upload button we compres\rotate\crop images and put them on the server 

and for users, it like a prepare step before uploading