react-native-camera/react-native-camera #3011: I cannot capture a screenshot of other objects I put on the camera object.

aytacabay, 2020-10-27:
# Question
I have a question ?

- When the camera is on, I place another object on it. I give the object's style property position: 'absolute'. Then when I take a screenshot, the object is not visible.
- But when I cancel the camera object and use position: 'absolute' with another object, the screenshot captures it.

How can I solve this problem?

I am making software as a hobby. Sorry if this is logically wrong.
How do you think I should do this ?

doranteseduardo, 2020-10-27:
Can you put an example of your code?

aytacabay, 2020-10-27:
import React, { useRef, useCallback, useState, useEffect } from 'react';
import { View, Text, TouchableOpacity, Image, ScrollView } from 'react-native';

import { RNCamera } from 'react-native-camera';
import ViewShot from 'react-native-view-shot';
import Draggable from 'react-native-draggable';

export default function CameraFuncScreenShot() {

    const [base64ScreenShot, setbase64ScreenShot] = useState(null)
    const [cameraIsOpen, setCameraIsOpen] = useState(true);

    let viewRef = useRef();

    let takeAPicture = async () => {
        if (cameraIsOpen) {
            await viewRef.current.capture().then(base64 => {
                setbase64ScreenShot(base64)
            })
        }
        await setCameraIsOpen(false)
    }

    useEffect(() => { }, [cameraIsOpen])

    return (
        <ViewShot
            ref={viewRef}
            options={{ format: 'jpg', quality: 0.9, result: 'base64' }}
            style={{ width: '100%', height: '100%', flexDirection: 'column', justifyContent: 'flex-start', alignItems: 'center' }}
        >
            <Draggable
                imageSource={require('../assets/imgIcons/1.png')}
                renderSize={80}
                x={200}
                y={300}
                // onDragRelease={this._changeFace}
                onLongPress={() => console.log('long press')}
                onShortPressRelease={() => console.log('press drag')}
                onPressIn={() => console.log('in press')}
                onPressOut={() => console.log('out press')}
            />
            {
                cameraIsOpen
                    ?
                    <RNCamera
                        options={{ format: 'jpg', quality: 0.9, result: 'base64' }}
                        style={{ width: '100%', height: '100%', flexDirection: 'column', justifyContent: 'flex-end', alignItems: 'center', backgroundColor: 'red', }}
                        type={RNCamera.Constants.Type.back}
                        flashMode={RNCamera.Constants.FlashMode.on}
                        zoom={0}
                        maxZoom={0}
                        androidCameraPermissionOptions={{
                            title: 'Permission to use camera',
                            message: 'We need your permission to use your camera',
                            buttonPositive: 'Ok',
                            buttonNegative: 'Cancel',
                        }}
                        androidRecordAudioPermissionOptions={{
                            title: 'Permission to use audio recording',
                            message: 'We need your permission to use your audio',
                            buttonPositive: 'Ok',
                            buttonNegative: 'Cancel',
                        }}
                    >
                        {
                            ({ camera, status, recordAudioPermissionStatus }) => {
                                if (status !== 'READY') {
                                    return (
                                        <View
[... 41 lines cut ...]
                        }
                    </RNCamera>
                    :
                    base64ScreenShot
                        ?
                        <View style={{ width: '100%', height: '100%', flexDirection: 'column', justifyContent: 'center', position: 'relative' }}>

                            <Image style={{ width: '100%', height: '100%', backgroundColor: '#EBE8E7' }} source={{ uri: `data:image/jpg;base64,${base64ScreenShot}` }} />

                            <View style={{ width: '100%', position: 'absolute', bottom: 10, flexDirection: 'row', justifyContent: 'space-around' }}>
                                <TouchableOpacity style={{ width: 40, justifyContent: 'center', }} onPress={() => { setTimeout(() => { setCameraIsOpen(true) }, 1000) }}>
                                    <Image style={{ width: 40, height: 40 }} source={require('../assets/img/cancel.png')} />
                                </TouchableOpacity>
                                <TouchableOpacity style={{ width: 40, alignSelf: 'center', }} onPress={async () => { console.log('Resim okeylendi..') }}>
                                    <Image style={{ width: 40, height: 40 }} source={require('../assets/img/check.png')} />
                                </TouchableOpacity>
                            </View>

                        </View>
                        :
                        <></>
            }
        </ViewShot >
    )
}

aytacabay, 2020-10-27:
> Can you put an example of your code?

import React, { useRef, useCallback, useState, useEffect } from 'react';
import { View, Text, TouchableOpacity, Image, ScrollView } from 'react-native';

import { RNCamera } from 'react-native-camera';
import ViewShot from 'react-native-view-shot';
import Draggable from 'react-native-draggable';

export default function CameraFuncScreenShot() {

    const [base64ScreenShot, setbase64ScreenShot] = useState(null)
    const [cameraIsOpen, setCameraIsOpen] = useState(true);

    let viewRef = useRef();

    let takeAPicture = async () => {
        if (cameraIsOpen) {
            await viewRef.current.capture().then(base64 => {
                setbase64ScreenShot(base64)
            })
        }
        await setCameraIsOpen(false)
    }
    useEffect(() => { }, [cameraIsOpen])

    return (
        <ViewShot
            ref={viewRef}
            options={{ format: 'jpg', quality: 0.9, result: 'base64' }}
            style={{ width: '100%', height: '100%', flexDirection: 'column', justifyContent: 'flex-start', alignItems: 'center' }}
        >
            <Draggable
                imageSource={require('../assets/imgIcons/1.png')}
                renderSize={80}
                x={200}
                y={300}
            />
            {
                cameraIsOpen
                    ?
                    <RNCamera
                        options={{ format: 'jpg', quality: 0.9, result: 'base64' }}
                        style={{ width: '100%', height: '100%', flexDirection: 'column', justifyContent: 'flex-end', alignItems: 'center', backgroundColor: 'red', }}
                        type={RNCamera.Constants.Type.back}
                        flashMode={RNCamera.Constants.FlashMode.on}
                        zoom={0}
                        maxZoom={0}
                        androidCameraPermissionOptions={{
                            title: 'Permission to use camera',
                            message: 'We need your permission to use your camera',
                            buttonPositive: 'Ok',
                            buttonNegative: 'Cancel',
                        }}
                        androidRecordAudioPermissionOptions={{
                            title: 'Permission to use audio recording',
                            message: 'We need your permission to use your audio',
                            buttonPositive: 'Ok',
                            buttonNegative: 'Cancel',
                        }}
                    >
                        {
                            ({ camera, status, recordAudioPermissionStatus }) => {
                                if (status !== 'READY') {
                                    return (
                                        <View
                                            style={{
                                                width: '100%',
                                                height: '100%',
                                                backgroundColor: 'lightgreen',
[... 37 lines cut ...]
                        }
                    </RNCamera>
                    :
                    base64ScreenShot
                        ?
                        <View style={{ width: '100%', height: '100%', flexDirection: 'column', justifyContent: 'center', position: 'relative' }}>

                            <Image style={{ width: '100%', height: '100%', backgroundColor: '#EBE8E7' }} source={{ uri: `data:image/jpg;base64,${base64ScreenShot}` }} />

                            <View style={{ width: '100%', position: 'absolute', bottom: 10, flexDirection: 'row', justifyContent: 'space-around' }}>
                                <TouchableOpacity style={{ width: 40, justifyContent: 'center', }} onPress={() => { setTimeout(() => { setCameraIsOpen(true) }, 1000) }}>
                                    <Image style={{ width: 40, height: 40 }} source={require('../assets/img/cancel.png')} />
                                </TouchableOpacity>
                                <TouchableOpacity style={{ width: 40, alignSelf: 'center', }} onPress={async () => { console.log('Resim okeylendi..') }}>
                                    <Image style={{ width: 40, height: 40 }} source={require('../assets/img/check.png')} />
                                </TouchableOpacity>
                            </View>

                        </View>
                        :
                        <></>
            }
        </ViewShot >
    )
}

doranteseduardo, 2020-10-27:
> import React, { useRef, useCallback, useState, useEffect } from 'react';
> import { View, Text, TouchableOpacity, Image, ScrollView } from 'react-native';
>
> import { RNCamera } from [...]

Ah, I see.
Well, this is not a react-native-camera issue, it's more related to how the ViewShot package works.
It has a compatibility table on its GitHub repo, I suggest you to check it out.

aytacabay, 2020-10-27:
### In this example, I don't use camera object. i put another text object instead and i can capture that.

<ViewShot
ref={viewRef}
options={{ format: 'jpg', quality: 0.9, result: 'base64' }}
style={{ width: '100%', height: '100%', flexDirection: 'column', justifyContent: 'flex-start', alignItems: 'center' }}
>
<View style={{ width: '100%', height: '30%' }}>
    <TouchableOpacity style={{ width: 65, alignSelf: 'center', }} onPress={() => { takeAPicture() }}>
        <Image style={{ width: 65, height: '100%' }} source={require('../assets/img/dot-2.png')} />
    </TouchableOpacity>
</View>
<Text style={{ position: 'absolute', top: 100, left: 100, color: 'red' }}>DEDE</Text>

{
    base64ScreenShot
        ?
        <View style={{ width: '100%', height: '100%', flexDirection: 'column', justifyContent: 'center', position: 'relative' }}>

            <Image style={{ width: '100%', height: '100%', backgroundColor: '#EBE8E7' }} source={{ uri: `data:image/jpg;base64,${base64ScreenShot}` }} />

            <View style={{ width: '100%', position: 'absolute', bottom: 10, flexDirection: 'row', justifyContent: 'space-around' }}>
                <TouchableOpacity style={{ width: 40, justifyContent: 'center', }} onPress={() => { setTimeout(() => { setCameraIsOpen(true) }, 1000) }}>
                    <Image style={{ width: 40, height: 40 }} source={require('../assets/img/cancel.png')} />
                </TouchableOpacity>
                <TouchableOpacity style={{ width: 40, alignSelf: 'center', }} onPress={async () => { console.log('Resim okeylendi..') }}>
                    <Image style={{ width: 40, height: 40 }} source={require('../assets/img/check.png')} />
                </TouchableOpacity>
            </View>

        </View>
        :
        null
}

</ViewShot >

closed 2021-01-13 by stale[bot]